chariotsolutions/phonegap-nfc #162: cann't trigger the callback function ,and the addNdefListener return success.

olivelihui, 2015-01-05:
nfc.addNdefListener(
    writeTag,
    function() {
        alert("Success.");
    },
    function() {
        alert("Fail.");
    }
);

function writeTag(nfcEvent) {
    alert("write message!");
}

don, 2015-01-06:
@olivelihui I'm assuming you're running on Android? `addNdefListener` will only trigger a callback if you scan a NFC tag with an NDEF message. If your NFC tag is not NDEF or hasn't been formatted you'll need to use `addTagDiscoveredListener` or `addNdefFormatableListener`.

Also, if you have a phone with the Broadcom NFC chipset it won't be able to read Mifare Classic NFC tags even if they are NDEF formatted.

Try using another app like [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite) to determine what's on your NFC tags. You can use [NFC TagWriter by NXP](https://play.google.com/store/apps/details?id=com.nxp.nfc.tagwriter) to format and write messages to tags.

don, 2015-01-23:
Assuming you've fixed this by adding additional listeners. If not, reopen the issue.

closed 2015-01-23 by don