chariotsolutions/phonegap-nfc #163: Null error

armaldio, 2015-01-06:
Hello,
I successfully be able to read a tag but when I try to share, write or erase all that i got is a null error.
What does it mean and how to solve it ?
Am I going the wrong way ?

armaldio, 2015-01-07:
Great, i learn something cool ^^
Is there a way to filter results ?

armaldio, 2015-01-07:
[image]

This is the log, filtering with "nfc" while scanning a tag

armaldio, 2015-01-07:
Ok, i'm sorry, i solved the problem by removing alerts
It cause trouble in using nfc and probably break the functions
So don't display alerts while using nfc ^^
Thank you for you help and your attention

closed 2015-01-07 by armaldio

don, 2015-01-07:
@armaldio I'm glad you got this working

clintgh, 2015-12-07:
I encountered this issue once, and one reason is because I immediately removed or distanced the device from the NFC sticker before it succeeds/finishes the writing process.