chariotsolutions/phonegap-nfc #168: Windows 8.1 support?

drsouly, 2015-01-26:
First off all, thanks for your hard work!

Is there a way to support the windows platform?

Cheers

don, 2015-01-26:
Windows 8? Do you have a computer/tablet with a built in NFC reader or an external reader?

drsouly, 2015-01-26:
I'm currently evaluating the possibility of using NFC with windows tablets. According to this slightly outdated list, there should be windows devices with NFC.

http://en.wikipedia.org/wiki/List_of_NFC-enabled_mobile_devices

We'll wait buying such a device until we know if its theoretically possible.

Thanks for your quick response.

don, 2015-01-26:
I think the Windows 8 works well with Cordova and the Proximity APIs should be similar to the Windows Phone APIs.  This should be possible. (disclaimer: I'm not a windows developer.)  It would be easier to develop if the proximity apis work external NFC readers.

don, 2015-04-08:
See https://github.com/chariotsolutions/phonegap-nfc/pull/182

don, 2015-04-16:
@jlkalberer added support for Windows 8.1 in #182. This is available in phonegap-nfc 0.6.2

closed 2015-04-16 by don