chariotsolutions/phonegap-nfc #169: nfc.enabled() not firing when NFC is enabled

Faustxx, 2015-02-03:
First of all, thanks for your great work.

My problem is that nfc.enabled() is not working correctly for me.

I'm getting NO_NFC in devices without NFC, and NFC_DISABLED when this feature is manually disabled in the device. This is working as intended.

But neither of the callbacks (success/failure) are being called at all when NFC is enabled on the device.

I can read NFC tags without problem, I just don't get the NFC_OK message I should be getting.

I'm using:
Cordova 4.2.0
NFC 0.6.1 (retrieved with this command: cordova plugin add com.chariotsolutions.nfc.plugin")
Sony Xperia M

don, 2015-02-03:
I'll take a look at this soon

don, 2015-02-04:
Install the plugin from github until 0.6.2 is released

    cordova plugin add https://github.com/chariotsolutions/phonegap-nfc

closed 2015-02-04 by don

Faustxx, 2015-02-04:
It is working now.

Thanks a lot!

commit 9b8ccbd9 (2015-02-04): Update NfcPlugin.java

Fixes #169