chariotsolutions/phonegap-nfc #170: Error "Class not found" addNdefListener

allannaranjo, 2015-02-13:
Hi I'm getting an error when I try to add the listener (addNdefListener).
I'm using:
phonegap version: 4.2.0-0.24.2
Samsung Galaxy S3 with Android 4.4.2

nfc.addNdefListener(app.onNFC, function(){
	alert("Listener added...");
}, function(e){
	alert("Error adding listener added..." + JSON.stringify(e));
});

Any ideas on how to fix this?

NFC is enabled in the phone, I even hear a sound when sliding a NFC Card on the back of the phone.

don, 2015-02-13:
Which class isn't found? Can you paste the full stack trace?

How are you installing the plugin?

allannaranjo, 2015-02-13:
All I get when I do this:
alert("Error adding listener added..." + JSON.stringify(e));
is "Class not found".
When I try to get the exception stack (e.stack) I get undefined.
I installed the plugin via phonegap
phonegap plugin add com.chariotsolutions.nfc.plugin

don, 2015-02-13:
See if you can get the Android stack trace by running "adb logcat". I'm guessing the plugin didn't install and it can't find the p

allannaranjo, 2015-02-13:
Here it is, I hope this helps:

/DatabaseUtils(  532): 	at com.android.server.am.ActivityManagerService.handleIncomingUser(ActivityManagerService.java:15031)
E/DatabaseUtils(  532): 	at android.app.ActivityManager.handleIncomingUser(ActivityManager.java:2488)
E/DatabaseUtils(  532): 	at com.android.providers.settings.SettingsProvider.call(SettingsProvider.java:688)
E/DatabaseUtils(  532): 	at android.content.ContentProvider$Transport.call(ContentProvider.java:325)
E/DatabaseUtils(  532): 	at android.content.ContentProviderNative.onTransact(ContentProviderNative.java:275)
E/DatabaseUtils(  532): 	at android.os.Binder.execTransact(Binder.java:404)
E/DatabaseUtils(  532): 	at dalvik.system.NativeStart.run(Native Method)
W/ActivityManager(  532): Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL
D/PluginManager(18718): exec() call to unknown plugin: NfcPlugin
I/SpenGestureManager(  532): setFocusWindow21010
D/KeyguardUpdateMonitor( 1053): sendKeyguardVisibilityChanged(true)
D/KeyguardUpdateMonitor( 1053): handleKeyguardVisibilityChanged(1)
D/PointerIcon(  532): setMouseIconStyle1 pointerType: 1001iconType:101 flag:0
D/PointerIcon(  532): setMouseCustomIcon IconType is same.101
D/CordovaLog(18718): file:///android_asset/www/plugins/com.chariotsolutions.nfc.plugin/www/phonegap-nfc.js: Line 21 : Failed to initialize the NfcPlugin Class not found
I/chromium(18718): [INFO:CONSOLE(21)] "Failed to initialize the NfcPlugin Class not found", source: file:///android_asset/www/plugins/com.chariotsolutions.nfc.plugin/www/phonegap-nfc.js (21)
D/CordovaActivity(18718): onMessage(spinner,stop)
E/ViewRootImpl(18718): sendUserActionEvent() mView == null
W/InputMethodManagerService(  532): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@421e88a8 attribute=null, token = android.os.BinderProxy@42b14c08
D/KeyguardUpdateMonitor( 1053): sendKeyguardVisibilityChanged(true)
D/KeyguardUpdateMonitor( 1053): handleKeyguardVisibilityChanged(1)
I/SurfaceFlinger(  175): id=450 Removed DordovaApp (9/12)
I/SurfaceFlinger(  175): id=450 Removed DordovaApp (-2/12)

don, 2015-02-13:
Make sure you see NfcPlugin.java in platforms/android/src/...

If not remove and reinstall the plugin.  

If the src file is there, maybe the problem is related to "W/ActivityManager( 532): Permission Denial: get/set setting for user asks to run as user -2 but is calling from user 0; this requires android.permission.INTERACT_ACROSS_USERS_FULL"

For that, switch the user or add the permission to the manifest.

allannaranjo, 2015-02-13:
I'll try that, thanks Don

allannaranjo, 2015-02-13:
The NfcPlugin.java is there,
Here's my AndroidManifest.xml
Am I makinf the permissions change in the right place?
Thanks.

<?xml version='1.0' encoding='utf-8'?>
<manifest android:hardwareAccelerated="true" android:versionCode="1" android:versionName="0.0.1" package="com.greencube.nfc" xmlns:android="http://schemas.android.com/apk/res/android">
    <supports-screens android:anyDensity="true" android:largeScreens="true" android:normalScreens="true" android:resizeable="true" android:smallScreens="true" android:xlargeScreens="true" />
    <uses-permission android:name="android.permission.INTERNET" />
    <application android:hardwareAccelerated="true" android:icon="@drawable/icon" android:label="@string/app_name">
        <activity android:configChanges="orientation|keyboardHidden|keyboard|screenSize|locale" android:label="@string/activity_name" android:launchMode="singleTop" android:name="CordovaApp" android:theme="@android:style/Theme.Black.NoTitleBar" android:windowSoftInputMode="adjustResize">
            <intent-filter android:label="@string/launcher_name">
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>
    </application>
    <uses-sdk android:minSdkVersion="10" android:targetSdkVersion="19" />
    -<uses-permission android:name="android.permission.NFC" />
     +<uses-permission android:name="android.permission.INTERACT_ACROSS_USERS_FULL" />
    <uses-feature android:name="android.hardware.nfc" android:required="false" />
</manifest>

allannaranjo, 2015-02-13:
The NfcPlugin.java is there,
Here's my AndroidManifest.xml
Am I makinf the permissions change in the right place?
Thanks.

<?xml version='1.0' encoding='utf-8'?>
<manifest android:hardwareAccelerated="true" android:versionCode="1"
android:versionName="0.0.1" package="com.greencube.nfc" xmlns:android="
http://schemas.android.com/apk/res/android">
    <supports-screens android:anyDensity="true" android:largeScreens="true"
android:normalScreens="true" android:resizeable="true"
android:smallScreens="true" android:xlargeScreens="true" />
    <uses-permission android:name="android.permission.INTERNET" />
    <application android:hardwareAccelerated="true"
android:icon="@drawable/icon" android:label="@string/app_name">
        <activity
android:configChanges="orientation|keyboardHidden|keyboard|screenSize|locale"
android:label="@string/activity_name" android:launchMode="singleTop"
android:name="CordovaApp"
android:theme="@android:style/Theme.Black.NoTitleBar"
android:windowSoftInputMode="adjustResize">
            <intent-filter android:label="@string/launcher_name">
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>
        </activity>
    </application>
    <uses-sdk android:minSdkVersion="10" android:targetSdkVersion="19" />
    -<uses-permission android:name="android.permission.NFC" />
     +<uses-permission
android:name="android.permission.INTERACT_ACROSS_USERS_FULL" />
    <uses-feature android:name="android.hardware.nfc"
android:required="false" />
</manifest>

allannaranjo, 2015-02-13:
According to:
http://gotoanswer.stanford.edu/?q=Permission+Denial%3A+this+requires+android.permission.INTERACT_ACROSS_USERS_FULL
I cannot set that permission, one thing I noticed is that I have to call "cordova run android --device" using sudo.
Otherwise I get: Error: EACCES, permission denied '/Users/allannaranjo/sites/nfc/plugins/android.json'
Can this be the cause of the issue?

allannaranjo, 2015-02-14:
I deleted the project and recreated everything, still the same error..

Allan

allannaranjo, 2015-02-14:
I just pushed it to a repo, maybe you can take a look at it:

https://github.com/allannaranjo/nfc

Thanks Don.

don, 2015-02-14:
Your test project ran on my machine. OS X 10.10.2, Cordova 4.2.0, Latest Android SDK (Tool 24.0.2, Platform Tools 21, Build Tools 21.1.2). Phone is Nexus 5 with Android 5.0.1.

    git clone https://github.com/allannaranjo/nfc
    cd nfc
    rm platforms/android/local.properties
    rm platforms/android/CordovaLib/local.properties
    cordova run android

What does your environment look like? You mentioned running cordova with sudo, maybe root can't read some of the files?

[image]

allannaranjo, 2015-02-14:
Hi Don,
Thanks for testing it.

My environment is:
OSX 10.9.5
Cordova 3.4.0-0.1.3
Latest Android SDK (Tool 24.0.2, Platform Tools 21, Build Tools 21.1.2).
Phone is Samsung Galaxy S3, Android 4.2.2

Do you think the Cordova version can make the difference?, or the Android
version?

Regards,

Allan

allannaranjo, 2015-02-14:
Hi Don,

I upgraded cordova and it worked!!!..
Very excited!.

Thanks a lot for helping me on this.
Have a great day!

Allan

don, 2015-02-14:
@allannaranjo Great news. Glad it's working.

closed 2015-02-14 by don

allannaranjo, 2015-02-14:
Well, I guess I spoke ahead of time, it does not work yet.

I see the alert "NFC Ready", however when I pass a tag, the device makes a
sound, but the onNFC never fires.

I pushed the new code.

Allan

allannaranjo, 2015-02-14:
Just a bit more info on this:

The tag I'm reading is a MiFare 1K Card 13.56MHz #359

Not sure if that helps

Allan

allannaranjo, 2015-02-14:
Here are some logs from adb logcat:

D/NativeNfcTag( 1181): Check NDEF Failed - status = 255
D/NativeNfcTag( 1181): Check NDEF Failed - status = 255
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 0, format 0, channelMask 3, flags 0
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 0, format 0, channelMask 3, flags 0
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 44100, format 1, channelMask 3, flags 4
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): startOutput() output 2, stream 5, session
214
V/AudioPolicyManagerBase(  177): changeRefCount() stream 5, count 2
W/AudioFlinger(  177): session id 213 not found for pid 1181
I/ServiceKeeper(  751): In getseinfo pid = 751 uid = 1000 seinfo= system
I/ServiceKeeper(  751): In getseinfo pid = 751 uid = 1000 seinfo= system
V/AudioPolicyManagerBase(  177): stopOutput() output 2, stream 5, session
213
V/AudioPolicyManagerBase(  177): changeRefCount() stream 5, count 1
V/AudioPolicyManagerBase(  177): releaseOutput() 2
D/NativeNfcTag( 1181): Tag lost, restarting polling loop
E/Sensors (  751): accelHandler -0.269440 8.183782 4.557722
D/SSRMv2:SIOP(  751): SIOP:: AP = 400, Delta = 10
V/AudioPolicyManagerBase(  177): stopOutput() output 2, stream 5, session
214
V/AudioPolicyManagerBase(  177): changeRefCount() stream 5, count 0
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
V/AudioPolicyManagerBase(  177): getNewDevice() selected device 0
V/AudioPolicyManagerBase(  177): setOutputDevice() output 2 device 0000
force 0 delayMs 192
V/AudioPolicyManagerBase(  177): setOutputDevice() prevDevice 0002
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
V/AudioPolicyManagerBase(  177): setOutputDevice() setting same device 0000
or null device for output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 0, format 0, channelMask 3, flags 0
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 0, format 0, channelMask 3, flags 0
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 44100, format 1, channelMask 3, flags 4
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): startOutput() output 2, stream 5, session
215
V/AudioPolicyManagerBase(  177): changeRefCount() stream 5, count 1
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
W/AudioPolicyManagerBase(  177): stream type [13], return media strategy
V/AudioPolicyManagerBase(  177): getNewDevice() selected device 2
V/AudioPolicyManagerBase(  177): setOutputDevice() output 2 device 0002
force 0 delayMs 0
V/AudioPolicyManagerBase(  177): setOutputDevice() prevDevice 0002
V/AudioPolicyManagerBase(  177): setOutputDevice() setting same device 0002
or null device for output 2
V/AudioPolicyManagerBase(  177): releaseOutput() 2
[... 13 lines cut ...]
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 0, format 0, channelMask 3, flags 0
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
V/AudioPolicyManagerBase(  177): getOutput() device 2, stream 5,
samplingRate 44100, format 1, channelMask 3, flags 4
V/AudioPolicyManagerBase(  177): getOutputsForDevice device 0002 -> 0002
V/AudioPolicyManagerBase(  177): getOutput() returns output 2
D/SensorService(  751):  [AR] -0.8 9.0 4.5
D/SensorService(  751): AutoRotationSensor::process: Ar_SensorChanged
oldrotation = [255], rotation = [0]
V/AudioPolicyManagerBase(  177): startOutput() output 2, stream 5, session
216
V/WindowOrientationListener(  751): Rotation Sensor : x=0.0
V/AudioPolicyManagerBase(  177): changeRefCount() stream 5, count 2
W/AudioFlinger(  177): session id 215 not found for pid 1181
V/AudioPolicyManagerBase(  177): stopOutput() output 2, stream 5, session
215
V/AudioPolicyManagerBase(  177): changeRefCount() stream 5, count 1
V/AudioPolicyManagerBase(  177): releaseOutput() 2
I/ServiceKeeper(  751): In getseinfo pid = 751 uid = 1000 seinfo= system
I/ServiceKeeper(  751): In getseinfo pid = 751 uid = 1000 seinfo= system
I/ServiceKeeper(  751): In getseinfo pid = 751 uid = 1000 seinfo= system
D/NativeNfcTag( 1181): Tag lost, restarting polling loop

allannaranjo, 2015-02-14:
Issues I see:

D/NativeNfcTag( 1181): Check NDEF Failed - status = 255 and
D/NativeNfcTag( 1181): Tag lost, restarting polling loop

Any ideas?

allannaranjo, 2015-02-14:
I used the NFC TagInfo app you used in the video:
https://www.youtube.com/watch?v=iOBvE8Wn4Cc#t=2746

And I get the tag info, but the phonegap app never reads my tag.

Allan

don, 2015-02-14:
I'm guessing that your Mifare Classic tag is **not formatted** as NDEF and you're using [nfc.addNdefListener](). In this case Android see the tag, but your app won't get the data since you're asking for NDEF info.

You can use [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en) to read the NFC tag to verify this assumption. In that case [nfc.addTagDiscoveredListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddtagdiscoveredlistener) will read the NFC tag id. Alternatively you can use [NXP Tag Writer](https://play.google.com/store/apps/details?id=com.nxp.nfc.tagwriter&hl=en) to write an NDEF message on that tag.

Note that newer phones with the Broadcom NFC chipsets will **not** be able to read Mifare Classic tags. 
Your S3 has the NXP chipset so you should be OK.

 You should get our [NFC book](http://shop.oreilly.com/product/0636920021193.do) ;)

allannaranjo, 2015-02-14:
Hi,

Correct, when used nfc.addTagDiscoveredListener and place the tag, I get an
alert when reading.

I will for sure get my reading started...

Thanks..

Allan

mork80, 2015-02-15:
New devices like Nexus 5 or Galaxy S4/5 have new reader chipset, Mifare 1k
are not supported anymore. Only NTAG tags are fully supported. My Nexus 5
can read only the serial of Mifare but cannot get NDEF records, even if I
have formatted them with my Nexus 4.