chariotsolutions/phonegap-nfc #172: plugin doesn't share with latest BB10.3.1.2243

firefoxpdm, 2015-02-21:
Hi

We are trying to make a small app that would require p2p communication between phones. The plugin is working fine on Android 4.4.2, but we're having problems on BB10.

The first problem was that every time we tried to share data, the Evernote application popped up. This was finally solved by making a small change in the phonegap-nfc-blackberry.js file. Based on the information found on http://developer.blackberry.com/native/documentation/core/nfc.html we added an additional "target" : "sys.NFCViewer" parameter to the data passed to the blackberry.invoke. After that BB's NFC sharing screen appeared, however it displayed that the NFC content had no data in it, and subsequently showed a "Failed to prepare data to share." error message. (http://oi57.tinypic.com/2hebre8.jpg)

After this we checked out the https://github.com/don/phonegap-p2p demo application, hoping to figure out what the problem was, but it also wasn't working properly. When we clicked the "Share Message" checkbox a JS alert appeared with the text "Failed to share tag undefined."

Aside to these, another interesting behaviour was that when we attached the addNdefListener, both the error and the success handlers were executed.

The strange thing is, the plugin seems to be able to properly handle incoming messages, it only has an issue with sharing them. Also, it is a bit problematic that the success handler for the share is called right when the BB NFC screen is displayed.

Is there any chance you could have a look at these issues?

Cheers

JohnMcLear, 2015-02-21:
Any chance you can isolate your code down to a replicatable test case so we can attempt to emulate locally?

If I attempt to run the phonegap-p2p on a BB device it wont function as expected?

mosonyi, 2015-02-21:
I tried it as well. I had to delete the platforms/blackberry10 and add again to make it install.
(cordova platform add blackberry10)
The installation was success but it doesn't work as expected. It can receive message from an android device, but can't send anything. JS error:  "Failed to share tag undefined."

firefoxpdm, 2015-02-22:
There isn't much to isolate, really:

```javascript
'use strict';

app.controller('Controller', function() {
	$scope.triggerShare = function() {
		nfcHandler.share("payload");
	}
});

var nfcHandler = (function() {
	
	function share(message) {
		c.log("sending message '" + message + "'");

		var payload = [
            ndef.textRecord(message)
        ];
		c.log("a");
        nfc.share(payload, function() {
    		c.log("should stop sending message '" + message + "'.");
        }, function() {
        	c.log("nfc sharing failed...");
        });
		c.log("finalized sending message '" + message + "'");
	}
	
	return {
		share : share
	};
})();
```

This code part is responsible for sending out the message. It works fine on Android, but fails in BB10. But the functionality is wrong for the demo app as well:

[image]

don, 2015-02-24:
It looks like something changed with the BB10 API . I tried with BlackBerry WebWorks 2.2.0.15 and a Z10 with 10.2.1.3062

    var success = function() { console.log("success"); };
    var failure = function(reason) { console.log("error " + reason); };
    var message = [ ndef.textRecord("hello") ];
    nfc.share(message, success, failure);

[image]

If you get this working, submit a pull request.

don, 2015-02-24:
I was missing the invoke plugin

    cordova plugin add com.blackberry.invoke

Now I see the behavior that you mentioned.

don, 2015-02-24:
It's not clear what webworks wants for the data parameter. Previously the bytes were encoded in a string. 
See http://developer.blackberry.com/html5/documentation/v2_2/nfc.html "Sharing NDEF messages".

Hopefully the encoding can be adjusted to get this working again.

don, 2015-02-25:
@mosonyi  what do you mean by "The config.xml should be edited to add a valid author tag."? Do you have an example of what should be added?

don, 2015-02-25:
I can't get webworks to accept generic NDEF bytes or find any examples of code using `application/vnd.rim.nfc.ndef` to write generic NDEF messages. I'm not sure why this broke.

As a work around you could use blackberry invoke directly if you're only sending one text record or one uri record.

Plain Text 

    query = {
        "action": "bb.action.SHARE",
        "type": "text/plain",
        "data": "hello, world",
        "target": "sys.NFCViewer"
    };
    blackberry.invoke.invoke(query, success, failure);

URI

    query = {
        "action": "bb.action.SHARE",
        "uri": "http://cordova.io",
        "target": "sys.NFCViewer"
    };
    blackberry.invoke.invoke(query, success, failure);

@mosonyi @firefoxpdm If you find code examples or more info, please post here.

closed 2015-02-25 by don

don, 2015-02-25:
`blackberry.invoke.invoke` was encoding the data incorrectly. Encoding the data with `btoa` and using `cordova.exec` for `invoke` seems to work. There's still an issue with both callbacks being called #173

commit 9abdc73e (2015-02-25): update nfc.share for BB 10.2
fixed #172