chariotsolutions/phonegap-nfc #175: App crashes on startup when adding NFC plugin

Gil401, 2015-03-06:
Hi

This is the error I am getting from my adb logcat when trying to run theapp:

/PluginManager( 7534): init()
W/System.err( 7534): java.lang.ClassNotFoundException: com.chariotsolutions.nfc.plugin.NfcPlugin
W/System.err( 7534):    at java.lang.Class.classForName(Native Method)
W/System.err( 7534):    at java.lang.Class.forName(Class.java:251)
W/System.err( 7534):    at java.lang.Class.forName(Class.java:216)
W/System.err( 7534):    at org.apache.cordova.PluginManager.instantiatePlugin(PluginManager.java:403)
W/System.err( 7534):    at org.apache.cordova.PluginManager.getPlugin(PluginManager.java:193)
W/System.err( 7534):    at org.apache.cordova.PluginManager.startupPlugins(PluginManager.java:116)
W/System.err( 7534):    at org.apache.cordova.PluginManager.init(PluginManager.java:92)
W/System.err( 7534):    at org.apache.cordova.CordovaWebView.init(CordovaWebView.java:163)
W/System.err( 7534):    at org.apache.cordova.CordovaActivity.init(CordovaActivity.java:345)
W/System.err( 7534):    at org.apache.cordova.CordovaActivity.init(CordovaActivity.java:320)
W/System.err( 7534):    at io.cordova.hellocordova.MainActivity.onCreate(MainActivity.java:31)
W/System.err( 7534):    at android.app.Activity.performCreate(Activity.java:5231)
W/System.err( 7534):    at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)
W/System.err( 7534):    at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2227)
W/System.err( 7534):    at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2313)
W/System.err( 7534):    at android.app.ActivityThread.access$800(ActivityThread.java:147)
W/System.err( 7534):    at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1226)
W/System.err( 7534):    at android.os.Handler.dispatchMessage(Handler.java:102)
W/System.err( 7534):    at android.os.Looper.loop(Looper.java:136)
W/System.err( 7534):    at android.app.ActivityThread.main(ActivityThread.java:5137)
W/System.err( 7534):    at java.lang.reflect.Method.invokeNative(Native Method)
W/System.err( 7534):    at java.lang.reflect.Method.invoke(Method.java:515)
W/System.err( 7534):    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:801)
W/System.err( 7534):    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:617)
W/System.err( 7534):    at dalvik.system.NativeStart.main(Native Method)
W/System.err( 7534): Caused by: java.lang.NoClassDefFoundError: com/chariotsolutions/nfc/plugin/NfcPlugin
W/System.err( 7534):    ... 25 more
W/System.err( 7534): Caused by: java.lang.ClassNotFoundException: Didn't find class "com.chariotsolutions.nfc.plugin.NfcPlugin" on path: DexPathList[[zip file "/data/app/io.cordova.hell
ocordova-1.apk"],nativeLibraryDirectories=[/data/app-lib/io.cordova.hellocordova-1, /vendor/lib, /system/lib]]
W/System.err( 7534):    at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
W/System.err( 7534):    at java.lang.ClassLoader.loadClass(ClassLoader.java:497)
W/System.err( 7534):    at java.lang.ClassLoader.loadClass(ClassLoader.java:457)
W/System.err( 7534):    ... 25 more
I/System.out( 7534): Error adding plugin com.chariotsolutions.nfc.plugin.NfcPlugin.
D/AndroidRuntime( 7534): Shutting down VM
W/dalvikvm( 7534): threadid=1: thread exiting with uncaught exception (group=0x4167fd58)
E/AndroidRuntime( 7534): FATAL EXCEPTION: main
E/AndroidRuntime( 7534): Process: io.cordova.hellocordova, PID: 7534
E/AndroidRuntime( 7534): java.lang.RuntimeException: Unable to start activity ComponentInfo{io.cordova.hellocordova/io.cordova.hellocordova.MainActivity}: java.lang.NullPointerException

E/AndroidRuntime( 7534):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2263)
E/AndroidRuntime( 7534):        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2313)
E/AndroidRuntime( 7534):        at android.app.ActivityThread.access$800(ActivityThread.java:147)
E/AndroidRuntime( 7534):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1226)
E/AndroidRuntime( 7534):        at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 7534):        at android.os.Looper.loop(Looper.java:136)
E/AndroidRuntime( 7534):        at android.app.ActivityThread.main(ActivityThread.java:5137)
E/AndroidRuntime( 7534):        at java.lang.reflect.Method.invokeNative(Native Method)
E/AndroidRuntime( 7534):        at java.lang.reflect.Method.invoke(Method.java:515)
E/AndroidRuntime( 7534):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:801)
E/AndroidRuntime( 7534):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:617)
E/AndroidRuntime( 7534):        at dalvik.system.NativeStart.main(Native Method)
E/AndroidRuntime( 7534): Caused by: java.lang.NullPointerException
E/AndroidRuntime( 7534):        at org.apache.cordova.PluginManager.getPlugin(PluginManager.java:195)
E/AndroidRuntime( 7534):        at org.apache.cordova.PluginManager.startupPlugins(PluginManager.java:116)
E/AndroidRuntime( 7534):        at org.apache.cordova.PluginManager.init(PluginManager.java:92)
E/AndroidRuntime( 7534):        at org.apache.cordova.CordovaWebView.init(CordovaWebView.java:163)
E/AndroidRuntime( 7534):        at org.apache.cordova.CordovaActivity.init(CordovaActivity.java:345)
E/AndroidRuntime( 7534):        at org.apache.cordova.CordovaActivity.init(CordovaActivity.java:320)
E/AndroidRuntime( 7534):        at io.cordova.hellocordova.MainActivity.onCreate(MainActivity.java:31)
E/AndroidRuntime( 7534):        at android.app.Activity.performCreate(Activity.java:5231)
E/AndroidRuntime( 7534):        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)
E/AndroidRuntime( 7534):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2227)
E/AndroidRuntime( 7534):        ... 11 more
W/ActivityManager( 1193):   Force finishing activity io.cordova.hellocordova/.MainActivity

Gil401, 2015-03-06:
I susspect it is related to the fact that i am using a few plugins... is there something needs to be changed in the android.json maybe?

Curretnly it looks like this:
{
    "prepare_queue": {
        "installed": [],
        "uninstalled": []
    },
    "config_munge": {
        "files": {
            "res/xml/config.xml": {
                "parents": {
                    "/widget": [
                        {
                            "xml": "<feature name=\"NfcPlugin\"><param name=\"android-package\" value=\"com.chariotsolutions.nfc.plugin.NfcPlugin\" /><param name=\"onload\" value=\"true\" /></feature>",
                            "count": 1
                        },
                        {
                            "xml": "<feature name=\"BluetoothSerial\"><param name=\"android-package\" value=\"com.megster.cordova.BluetoothSerial\" /></feature>",
                            "count": 1
                        }
                    ],
                    "/*": [
                        {
                            "xml": "<feature name=\"BluetoothLePlugin\"><param name=\"android-package\" value=\"com.randdusing.bluetoothle.BluetoothLePlugin\" /></feature>",
                            "count": 1
                        },
                        {
                            "xml": "<feature name=\"Device\"><param name=\"android-package\" value=\"org.apache.cordova.device.Device\" /></feature>",
                            "count": 1
                        },
                        {
                            "xml": "<feature name=\"SplashScreen\"><param name=\"android-package\" value=\"org.apache.cordova.splashscreen.SplashScreen\" /><param name=\"onload\" value=\"true\" /></feature>",
                            "count": 1
                        }
                    ]
                }
            },
            "AndroidManifest.xml": {
                "parents": {
                    "/manifest": [
                        {
                            "xml": "<uses-permission android:name=\"android.permission.NFC\" />",
                            "count": 1
                        },
                        {
                            "xml": "<uses-feature android:name=\"android.hardware.nfc\" android:required=\"false\" />",
                            "count": 1
                        },
                        {
                            "xml": "<uses-permission android:name=\"android.permission.BLUETOOTH\" />",
                            "count": 2
                        },
                        {
                            "xml": "<uses-permission android:name=\"android.permission.BLUETOOTH_ADMIN\" />",
                            "count": 2
                        }
                    ]
                }
            }
        }
    },
    "installed_plugins": {
        "com.chariotsolutions.nfc.plugin": {
            "PACKAGE_NAME": "com.gym8.app"
        },
        "com.megster.cordova.bluetoothserial": {
            "PACKAGE_NAME": "com.gym8.app"
        },
        "com.randdusing.bluetoothle": {
            "PACKAGE_NAME": "com.gym8.app"
        },
        "org.apache.cordova.console": {
            "PACKAGE_NAME": "com.gym8.app"
        },
        "org.apache.cordova.device": {
            "PACKAGE_NAME": "com.gym8.app"
        },
        "org.apache.cordova.splashscreen": {
            "PACKAGE_NAME": "com.gym8.app"
        }
    },
    "dependent_plugins": {}
}
thanks

don, 2015-03-16:
It looks like the problem is that the phonegap-nfc plugin is not installed.

     java.lang.ClassNotFoundException: com.chariotsolutions.nfc.plugin.NfcPlugin

You should never need to edit android.json. Use `cordova plugin add com.chariotsolutions.nfc.plugin` plugman should install all the files in the correct places. If `cordova plugin ls` shows the plugin installed, try removing and re-adding the plugin.

don, 2015-04-16:
@Gil401 did you solve this issue?

closed 2015-10-23 by don