chariotsolutions/phonegap-nfc #176: Password protection

frostyelkarne, 2015-03-14:
I have a tag that is password protected, how can I send the password?

I read the NTAG specs and found that there is a command called PWD_AUTH that should be used.
From the spec: "A protected memory area can be accessed only after a successful password verification using the PWD_AUTH command"

don, 2015-03-16:
@frostyelkarne this plugin concentrate on NDEF support for NFC. Right now there's no support for PWD_AUTH or NTAG.

lezardrouge, 2015-04-01:
Hello Don, 
Do you plan to implement it ? Otherwise, is there another way (plugin...) to do it anyway ?
Thanks

frostyelkarne, 2015-04-01:
I did a simple plugin for my own app that sends a pw and then read 200 bytes from memory. Only for Android and NTAG210 tags. https://github.com/FrostyElk/cordova-mifare

JohnMcLear, 2015-04-01:
Looks interesting @frostyelkarne , any intention to extend to other platforms or create a pull request with this functionality in?

frostyelkarne, 2015-04-01:
It uses the MIFARE SDK from NXP so it's restricted to Android, I haven't seen any other versions of that SDK.

don, 2015-04-02:
@frostyelkarne can your plugin be used with this plugin or does it work stand alone?

CalvinEng, 2015-07-08:
Hello Don, I need to develop an application that can read and write nfc tag with password protection, is there any intention to add protection into the plugin?

allannaranjo, 2015-07-08:
Hi,
I would recommend the use of "crypto".
Depending on your need, you can:
1. Create a secret that you control and use it to encrypt and decrypt the
info stored in the tag.
2. Allow the user to enter their own secret, with this method, only the
user can decrypt/encrypt the tag info

In any of the cases, if the tag can be decrypted with the provided secret
then you can write, if not, you can alert the user.

Good luck.

Allan

don, 2015-07-08:
@CalvinEng I don't have any plans to add this now. As @allannaranjo points out you can do this before reading and writing the NFC tag. You should also look at @frostyelkarne implementation of NTAG https://github.com/FrostyElk/cordova-mifare, you might be able to get it to work with this plugin.

clarklight, 2015-10-08:
@don @allannaranjo I am quite new to NFC on phonegap, just have a quick question regarding the "crypto" and password protection part. 

I am also looking at the NTAG 21x family. 
Can the "crypted" content still be read when it hasnt been decrypted yet?? (Read but encrypted?)
And the same for PWD_AUTH, can the content still be read(displayed) if the password hasnt been entered yet??

I am planning/trying to design a "anti-cloning" mechanism for my small test project( i know of the Original Signature for anti-cloning but totally have no idea how it works so i thought i can use the password protected area to achieve similar function) , that i hides the Account ID within the password protected area, and only when the password is entered, then the account ID will be revealed, so that specific area can not be cloned without the password.

don, 2015-10-08:
This plug handles NDEF. You could write an encrypted NDEF payload using this library, but anyone could read the encrypted data. There's nothing to stop people from cloning an NDEF message.

To access tag specific function that access password protected areas, you're going to need to use native android tag specific apis.

clarklight, 2015-10-08:
@don  thanks for the reply, would you by any chance know if the password protected areas is "hidden" or not before the password is entered, i know this plugin just handles NDEF, i was just wondering. 

I couldnt really find out about these info on the NTAG documents.

don, 2015-10-08:
It looks like the tag doesn't allow access to the protected areas until the password is entered. But they recommend encryption if you need a "higher level of protection".

See NTAG210_212.pdf [link](http://www.nxp.com/products/identification_and_security/smart_label_and_tag_ics/ntag/series/NTAG210_NTAG212.html)

> 8.7 Password verification protection
The memory write or read/write access to a configurable part of the memory can be constrained to a positive password verification. The 32-bit secret password (PWD) and the 16-bit password acknowledge (PACK) response are typically programmed into the configuration pages at the tag personalization stage.

>The AUTHLIM parameter specified in Section 8.5.7 can be used to limit the negative verification attempts.

>In the initial state of NTAG21x, password protection is disabled by a AUTH0 value of FFh. PWD and PACK are freely writable in this state. Access to the configuration pages and any part of the user memory can be restricted by setting AUTH0 to a page address within the available memory space. This page address is the first one protected.

>Remark: The password protection method provided in NTAG21x has to be intended as an easy and convenient way to prevent unauthorized memory accesses. If a higher level of protection is required, cryptographic methods can be implemented at application layer to increase overall system security.

clarklight, 2015-10-08:
@don  thanks so so much for the info, i also took a look on this NFC tag info app, frostyelkarne mentioned the NFC taginfo app could read the password protected area straight away, but i will have to research further. Thanks 
https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en

don, 2015-10-08:
@clarklight I haven't seen a NTAG21x android driver. For most stuff you could probably use [MifareUltralight](https://developer.android.com/reference/android/nfc/tech/MifareUltralight.html) to read and write pages. For the password protection you probably need to write commands directly to the tag using [transceive(byte[])](https://developer.android.com/reference/android/nfc/tech/NfcB.html#transceive(byte[])) from NfcB (or whatever the correct tag tech is.)

clarklight, 2015-10-08:
@don thanks for the info, i think i will have to give it more thought to the security structure to what i want to achieve before i dig into the code itself further..... i am a little hesitant as it seems to be quite easy to replicate the data from a card(to cause any server's action) maybe i will have to look into Desfire and other options.

 I will also look into @frostyelkarne's code https://github.com/FrostyElk/cordova-mifare
Just to check the degree of security and test the cloning(and reading password protected area) before i push too far down the line.  

Thanks again anyway!

clarklight, 2015-10-10:
@don I did further research seems like for NTAG 21x and Utlralight i can use their Mifare LITE SDK to write into the password area, your_object.programPWDPack(byte[] pwd, byte[] pack)
pwd = 4 bytes of password, pack = {0x00, 0x00}

enablePasswordProtection method: your_object.enablePasswordProtection(false, objnTag210.getFirstUserpage())

But i am not sure if i can do the same for Desfire with their 3DES/AES on the LITE SDK version. 

Just got 1 question tho,not sure if anyone got any experience in Ultralight EV1, if the security is same as Desfire EV1 or the NTAG 21x?

chompirax, 2016-06-23:
@don You are planning to upgrade this plugin with password protection?

don, 2016-06-23:
@chompirax it's on my list, but very low priority. It probably won't happen until I need it for client.

don, 2018-04-18:
Closing old issue. Will add this when a client needs it.

closed 2018-04-18 by don

ThirupathiRaoM, 2022-01-19:
hello Everybody 
I am Working on Core NFC For one of my iPhone application while I am trying to lock the NFC Tag using password protected I am not getting any Solution regarding that but my client given Some applications in those application password protected working fine please let us know if you find any solution regarding Password protected in Swift5 please share here it will help me a lot Thanks