chariotsolutions/phonegap-nfc #178: nfc.enabled()  Success callback is never called.

zachdj, 2015-03-25:
When using the nfc.enabled() function, the failure callback will be called if NFC is disabled.  However, if NFC is enabled, the success callback is not called.

don, 2015-03-25:
This was fixed in 9b8ccbd91a6afddfd3b647392c602d51757705c8 but hasn't been pushed to plugins.cordova.io. Can you try installing the plugin from github until I push another release?

    cordova plugin rm com.chariotsolutions.nfc.plugin
    cordova plugin add https://github.com/chariotsolutions/phonegap-nfc

zachdj, 2015-03-26:
Thanks!  I'll try that.  I appreciate the quick response.

didlich, 2015-03-31:
@don thanks for the fix,
I added this PR for enabled() some time ago, after that I updated to new versions of the NFC plugin, but saw erronous behavior in my app, I'm polling the enable() interface to show a message when the user should siwitch NFC on, when user disables the NFC and enables the NFC again the message did not disappear
then I saw that in version 0.6.1 the call back was never called, so I use the current master branch which fixed the erronous behavior of my app

I hope for a quick release of a new version

don, 2015-04-16:
Duplicate of #169. Fixed in 0.6.2.

closed 2015-04-16 by don