chariotsolutions/phonegap-nfc #179: Error occured in NfcPlugin.java

polim1022, 2015-03-30:
hi, i follow all steps to install the plugin but it happens that there is some error occurs in NfcPlugin.java. Error says that : the method sendJavascript(String) from the type CordovaWebView is deprecated. What does it mean? Sorry i'm just a newbie. I'm doing this for my project. Urgent to solve this problem. Can tell me how to solve this? Thanks a lot

[image]

don, 2015-03-30:
That deprecation method is a warning, not an error. The deprecated messages means that `sendJavascript` will be removed in a future version of Cordova. You can safely ignore the warning and your app will work fine.

polim1022, 2015-03-30:
Ya..you are right. I just run it and it is fine. Thanks for your fast reply!

closed 2015-04-08 by don