chariotsolutions/phonegap-nfc #181: write only one record on multiple records

lezardrouge, 2015-04-01:
Hello,
Is there a way to write only in a record without deleting the others already set ?
For example my payload contains a record with text and another with some other data, let's say an url ; I want to rewrite only the text (exact same length) and do not touch the url. 
I would like to avoid to rewrite everything because I have 8ko of data and it takes too much time (and it's also dangerous to rewrite this data).
Thank you

JohnMcLear, 2015-04-01:
You should do a read function, append to an object, then write...

don, 2015-04-02:
@lezardrouge the underlying APIs only write the whole NDEF message, so @JohnMcLear's read, append and write solution is the best way to do this.

lezardrouge, 2015-04-02:
Thank you both for your answers. 
@don when you talk about the underlying API, is it a limitation of Cordova or it would be the same problem with a native app ?

don, 2015-04-08:
@lezardrouge you'd have the same problem with native Android too. You need to write the complete NDEF message, you can't update an existing message.

closed 2015-04-16 by don