chariotsolutions/phonegap-nfc #183: Windows Phone 8.1 support

mletynski, 2015-04-06:
Hi,
Any plans for supporting windows phone 8.1 ?
Im trying to install nfc plugin via cordova CLI. I'm getting successful message that plugin for windows was installed however it does not have any effect (for e.g module.exports inside cordova_plugins.js is not filled) and same problem is with native c# classes and permissions.

I compared it to wp8 cordova platform and it has totally different project structure then windows cordova platform (it came with all windows platforms).

don, 2015-04-08:
I'd think the existing WP code *should* work on the newer phones, but this might not be the case. I'm trying to track down a newer windows phone so I can test https://github.com/chariotsolutions/phonegap-nfc/pull/182

mletynski, 2015-04-12:
Yes indeed it works with existing WP. I think its even better since such app will work with Windows Phone 8.0 and 8.1 where compiled with 8.1 will work only on that device. However it would be good to have it running via cordova CLI.

don, 2015-04-13:
It's taking a bit longer than anticipated to track down a Windows Phone 8.1 with NFC... Once I get one, I'll get an updated release that works with the CLI.

don, 2015-04-16:
Fixed with #182

closed 2015-04-16 by don