chariotsolutions/phonegap-nfc #184: Windows Phone 8.1 callback is not fired

mletynski, 2015-04-06:
I'm using this plugin for a long time in Android. Now it's time for Windows Phone. But it somehow does not work at all. I'm listening on NDEF tags (nfc.addNdefListener(callback, [onSuccess], [onFailure]);) However callback is not fired. Last thing which i see on debug console is:

Nfc Plugin
Registering for NDEF
The thread 0xfd4 has exited with code 259 (0x103).
'TaskHost.exe' (CoreCLR: Silverlight AppDomain): Loaded 'C:\windows\system32\System.Runtime.WindowsRuntime.ni.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
The thread 0x984 has exited with code 259 (0x103).

Ant thats all. I added some debug and i see that MessageReceivedHandler is invoked. However InvokeCustomScript method is deprecated. Maybe this is a problem. Any hint ? What i can check more ? I downloaded some native app for reading nfc tags and my tag was successfully read.

don, 2015-04-08:
InvokeCustomScript has been deprecated for a while, that shouldn't be the problem. Maybe the native API changed for 8.1? I'll try and track this down one I get a device.

mletynski, 2015-04-15:
After quick debug i saw that this.OnCustomScript is simply a null.

    [Obsolete]
        public void InvokeCustomScript(ScriptCallback script, bool removeHandler)
        {
            if (this.OnCustomScript != null)
            {
                this.OnCustomScript(this, script);
                if (removeHandler)
                {
                    this.OnCustomScript = null;
                }
            }
        }

I tried to invoke DispatchCommandResult(new PluginResult(PluginResult.Status.OK, JsonHelper.Serialize(tag)));

instead of InvokeCustomScript but i'm getting:

Failed to locate callback for id : NfcPlugin625794433

I will try to downgrade cordova to 3.6.x since now i'm using 4.3.0

mletynski, 2015-04-15:
Everything works with Cordova 3.6.3-0.2.13

don, 2015-04-16:
I merged in @jlkalberer patch for Windows Phone 8.1 #182 

Hopefully this fixes the issue with the latest Cordova. If not, reopen this issue.

closed 2015-04-16 by don