chariotsolutions/phonegap-nfc #185: A quick question / clarification

m-menon, 2015-04-26:
hi,

Firstly thanks a ton for the useful n nice plugin :-) I have this little problem and I think it is some tiny little thingy that I am missing somewhere or doing incorrectly. I running this code on device ready event

<pre>
nfc.addTagDiscoveredListener(
    onNfc,
    function () {
      // Do something
    },
    function (error) {
     // Handle errors if any
    }
);
</pre>

Now comes the somewhat tricky thing - I am using a foreground camera (because the default camera plugin kills / suspends the app the moment it loads up) plugin (https://github.com/shaithana/cordova-plugin-wezka-nativecamera). The plugin I mentioned runs the camera without kiiling cordova.

Now here's the problem: The moment even this 'foreground' camera loads up, and I try to scan a NFC card, it invokes the default NFC tag reader app instead of executing my onNfc event. The onNfc just read the card tag / id.

Here is onNfc event
<pre>
function onNfc(nfcEvent) {
    try {
        nfcScanResult = nfc.bytesToHexString(nfcEvent.tag.id);
        // Do something
    }
    catch(err) {
        nfcScanResult = "Error: " + err.message;
        // Handle the error gracefully
    }
}
</pre>

I am pretty new to Android and java. Any tip or suggestion that may help me solve this problem? Btw I have tried one of the suggestions which recommends 'addMimeTypeListener' to solve this problem. That doesn't work either. Can I know what exactly I would be required to add in the android manifest file as well.

<pre>
nfc.addMimeTypeListener(mimeType, callback, [onSuccess], [onFailure]);
</pre>

Thanks a ton,
M&M

don, 2015-04-29:
The NFC plugin uses foreground dispatching to read NFC tags. Once the camera launches (the built in camera or the plugin you use) NFC will be handled by background dispatching. In order to have your app handle NFC tags, you need to add an intent filter to the AndroidManifest.xml

Here's [one example](https://github.com/chariotsolutions/phonegap-nfc#launching-your-android-application-when-scanning-a-tag) the Android docs [have more](http://developer.android.com/guide/topics/connectivity/nfc/nfc.html#ndef-disc).

Once you have the intent-filter working you need the mime-type filter in your app. The mime-type doesn't matter for this use cases. (I usually put "" or "bogus".)  The plugin will fire a ndef-mime event, even when scanning a text, html or external tag from a background scan intent.

If you need help crafting the intent filter, post more details about the NFC tags you're trying to scan here.

m-menon, 2015-05-02:
hi Don,

Thanks a lot for the clarification. I have attached some information about the tags. Actually the problem that I have appears minuscule but it's taken me a few days and a good part of my wits. The preview camera that I run in one of my pages hangs the moment I scan the NFC tag. I believe the NFC reading part pauses (or pushes the app into background) for a fraction of a second and that causes the camera to freeze. I tried stopping the camera when the moment the App goes into background and resume it when the app comes back to foreground. This partially solves the problem but after 3-4 card scans my app crashes.

Thanks a lot for you time. 

Cheers,
M&M
[image]
[image]
[image]
[image]

don, 2015-05-03:
Since your tag is TNF Well Known RTD Text tag you'll want to filter for text tags.

    <intent-filter>
        <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
        <category android:name="android.intent.category.DEFAULT"/>
        <data android:mimeType="text/plain" />
    </intent-filter>

Yes, the intent filter should be in the main activity tag. [example](https://github.com/don/phonegap-nfc-reader/blob/master/platforms/android/AndroidManifest.xml#L1)

In `index.js` you'll need to add [nfc.addMimeTypeListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddmimetypelistener). You may be able to reuse the same callback you pass to [nfc.addNdefListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddndeflistener).

If you add an intent filter for TNF Well Known RTD Text your user will get prompted if there are multiple applications that handle this tag type. Most generic NFC programs will. If your app need exclusive access to the tag, consider a custom tag type. This can be done with a custom MIME type, or and EXTERNAL tag type.

I can't help much with the app crashing. Those problems can be tough. Fortunately it sounds like it's reproducible, which is good. I want the log files for where the error occurs then debug the Java code in IntelliJ to solve these type of problems.

m-menon, 2015-05-06:
hi Don,

Thanks a ton for the information. It has given me some leads and it seems like I am closer to solving the problem. If I still can't then I'll get the logs and get back to you.

Thanks once again. Really appreciate your time and the useful plugin.

Cheers,
M&M

closed 2015-05-08 by don