chariotsolutions/phonegap-nfc #187: Read the NFC tag when App opens

dukhanov, 2015-05-18:
Hello,

I noticed your answer on Stackoverflow: http://stackoverflow.com/a/30018182/4527512.
You said: if you want to read the NFC tag you should add a nfc.addMimeTypeListener to your code.

I have added this code in your method "parseMessage", in this place https://github.com/chariotsolutions/phonegap-nfc/blob/master/src/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java#L598, to see the message:

cordova.getActivity().runOnUiThread(new Runnable() {
                    public void run() {
                        Toast.makeText(cordova.getActivity().getApplication(), "Message: " + Util.tagToJSON(tag), Toast.LENGTH_LONG).show();
                    }
                });

This worked before and message appeared, but now, when the App opens the NFC tag, message is not appeared (appears empty message). (Maybe after my device was updated, this functionality is broken).

My NFC tag contains two messages:
1) the android app id (for opening the app)
2) some data which I want to read, when the app is opens

Could you help me?

Thanks.

P.S. 
I tested on Nexus 5. Android 5.1.

don, 2015-05-28:
You should add [nfc.addMimeTypeListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddmimetypelistener) to your JavaScript code. You don't need to modify the Java code.

dukhanov, 2015-05-29:
For the launching the App I use an Intent Filter:
<intent-filter>
    <action android:name="android.nfc.action.NDEF_DISCOVERED" />
    <category android:name="android.intent.category.DEFAULT" />
    <data android:scheme="http" android:host="example.com" android:pathPrefix="" />
</intent-filter>

1) If the app running and placed in the background, when the NFC tag is approached, the app opens and mime type event happens (the Android version). 
2) If application is not running, when the NFC tag is approached, the app opens and mime type event does not happens (the Android version). 

I think, when App is starting the onNewIntent event (Android) happens before the device ready event (Cordova).

I solved that problem by saving the Intent object, in the onNewIntent event. And when the device ready event happens, I am trying to get data from saved Intent.

P.S. tested on Nexus 5 (Android 5.1)

don, 2015-06-02:
The plugin supports launching the application from an intent when scanning an NFC tag. You shouldn't need to modify any code.

I created a sample project to demonstrate this behavior https://github.com/don/nfc-launch

There are cases when the intent arrives before the plugin is ready. This is why the plugin internally [saves a copy of the intent](https://github.com/chariotsolutions/phonegap-nfc/blob/master/src/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java#L711).

dukhanov, 2015-06-03:
Thanks for answer.

I found an issue. 
I use ionic and require js in my app.
Initialized the NfcPlugin event is fired before the device ready event is fired.

[image]

But now, I don't know how to solve that issue.

don, 2015-06-03:
@dukhanov I created a branch of my ionic-nfc-reader for you https://github.com/don/ionic-nfc-reader/tree/phonegap-nfc-187

This branch handles NFC tags from intents that launch the app.

Add your intent filter to the AndroidManifest.xml after adding the platform.

dukhanov, 2015-06-03:
I use require js for loading all JS files.
But "Initialized the NfcPlugin Event" happens before my event:
```js
// place where I want to subscribe on nfc events
require(['ionic', 'angular', 'app', 'routes'], function (ionic, angular, app) {
    document.addEventListener('deviceready', function () {

        console.log("main js run: " + (new Date()).getTime());
        angular.element().ready(function () {
            angular.resumeBootstrap([app['name']]);
        });               
        
    }, false);
});
```
Event: Initialized the NfcPlugin: 1433348808997 - timestamp
Event: main js run: 1433348809212 - timestamp
Event: Initialized the NfcSERVICE: 1433348809423 - timestamp

don, 2015-06-03:
I don't use require.js so I don't know how to fix this. Maybe try the forums (or irc) for ionic and require.js?

dukhanov, 2015-06-04:
Thanks a lot for help.

As I said before I solved that problem by saving the Intent object, in the onNewIntent event. 
And when the device ready event happens, I am trying to get NFC data from saved Intent.
This probably not the best solutions, but woks for me.

closed 2015-10-23 by don