chariotsolutions/phonegap-nfc #188: nfc plugin does not retain the last view app

pcalderoncity, 2015-06-01:
Hi, i have installed the pluggin but I have a problem. it works correctly when my phonegap app is showing but if I close it and I open it again does not keep the last view and the login page appears.

any ideas?

Thank you for your prompt reply.

allannaranjo, 2015-06-02:
I don't think this is related to the plugin, what do you mean by the
"login" page?
I believe you might be asking about restore the state of your application
which is unrelated to this plugin.
Allan

pcalderoncity, 2015-06-02:
Thanks for your replay Allan, Excuse me for my english is not so good. I will try to explain what is my problem. My phonegap app works correctly restoring the last view but when I install the nfc plugin and it reads the card if I close the app and i open it again the last view is missing and the app loses the last state. I think that i am doing something wrong. Can you help me with any idea? 

Thanks again.

don, 2015-06-02:
@pcalderoncity I agree with @allannaranjo that this sounds more like an application state problem than a problem with phonegap-nfc.

Do you see any errors from your app if you view the logs with  `adb logcat` or [device monitor](http://developer.android.com/tools/help/monitor.html)?
If you remove the plugin, does your application retain state when you exit and re-open it?
How does your application decide which page should be show? Are you overwriting this data after you read a nfc tag?

pcalderoncity, 2015-06-02:
Hi Don,
The logcat shows that my app calls again to the create method CordovaActivity.onCreate() and the life cicle starts again in my app login page. 

Reading only one card the app loses the state. I do not why?. Obiously if I do not put the card the problem does not appear. The app loses the state when it reads the first card.  My app is built with jquery using multi-page template, that is to say, multiple views in the same page.

For me is transparent the way that jquery or cordova handle the states, it works correctly by default.

I am new in the android world, I will appreciate any help.

Thanks Don.

don, 2015-06-02:
So the login page appears after you scan a NFC tag? If that's the case, you may be doing something that causes the page to navigate, reloading the app. Look at the JavaScript where you're processing the nfcEvent. Look for javascript errors at runtime. Run jshint on your code.

Also look at the log before you see CordovaActivity.onCreate() are there any errors? If you see a Java stack trace, it's possible the plugin is causing an issue. Send me the details.

If you share your code I can take a quick look.

pcalderoncity, 2015-06-02:
Thanks for your advice, I

 will run the jshint tool but it should be noted that I commented the js lines where it set the nfc listener, therefore I think my personal code is not the problem. Just installing the plugin without the listener and put the card for reading originates my bug.

Line 1298 
								//nfc.addTagDiscoveredListener(_onNfc,function (status){alert(">>>>" + status);},function (error) { CM_SUBE_BAJA_OBJ.showMessage(MSG_NFC_ERR + JSON.stringify(error));});

Thanks for your attention.

Below my code:

<!DOCTYPE html>
<html>
<head>
  	<meta name="viewport" content="initial-scale=1, minimum-scale=1, maximum-scale=1">
	<meta http-equiv="content-type" content="text/html;charset=ISO-8859-1" />
  	<title>SUBE-BAJA</title>
  	 
	<link rel="stylesheet" href="css/jquery-ui.min.css" >
 	<link rel="stylesheet" href="css/jquery.mobile-1.4.5.min.css">
  	<link rel="stylesheet" href="css/index.css">
  	<script src="js/jquery-1.11.0.min.js"></script>
 	<script src="js/jquery-ui.min.js"></script>
  	<script src="js/jquery.mobile-1.4.5.min.js"></script>
 
	<script type="text/javascript" src="cordova.js"></script> 
	

	<style>
	 .ui-header {
  height: 41px;   
}

.ui-autocomplete {
  position: absolute;
  top: 100%;
  left: 0;
  z-index: 10000;
  float: left;
  display: none;
  min-width: 160px;
  _width: 160px;
  padding: 4px 0;
  margin: 2px 0 0 0;
  list-style: none;
  background-color: #ffffff;
  border-color: #ccc;
  border-color: rgba(0, 0, 0, 0.2);
  border-style: solid;
  border-width: 1px;
  -webkit-border-radius: 5px;
  -moz-border-radius: 5px;
  border-radius: 5px;
  -webkit-box-shadow: 0 5px 10px rgba(0, 0, 0, 0.2);
  -moz-box-shadow: 0 5px 10px rgba(0, 0, 0, 0.2);
  box-shadow: 0 5px 10px rgba(0, 0, 0, 0.2);
  -webkit-background-clip: padding-box;
  -moz-background-clip: padding;
  background-clip: padding-box;
  *border-right-width: 2px;
  *border-bottom-width: 2px;

  .ui-menu-item > a.ui-corner-all {
    display: block;
    padding: 3px 15px;
    clear: both;
    font-weight: normal;
    line-height: 18px;
    color: #555555;
    white-space: nowrap;
[... 1308 lines cut ...]
	function enableTimer(data)
	{
		myService.registerForBootStart(data);
		if (data.TimerEnabled)
		{
			registerForUpdates(data);
	      
		}
		else
		{
			myService.enableTimer(60000, function(r){registerForUpdates(r)}, function(e){displayError(e)});
		}
	}
	
	function registerForUpdates(data)
	{
		if (!data.RegisteredForUpdates)
		{
		      myService.registerForUpdates(function(r){updateHandler(r)}, function(e){handleError(e)});
		}
	}
	// app.initialize();
	</script>
</body>
</html>

don, 2015-06-02:
Rather than pasting code here, a github repo or gist would be better.

pcalderoncity, 2015-06-02:
I am Sorry, the problem is not the plugin.  I think that it is jquery-cordova bug. Without the plugin occurs the same issue. 

Thank you very much for your time.

closed 2015-10-24 by don