chariotsolutions/phonegap-nfc #190: Cant get the NFC plugin to work on Android and Windows Phone

cmeegamarachchi, 2015-06-19:
I am trying to get build a test cordova application to read and write NFC tags targeting windows phones and android phone. When the test application is deployed to devices, on android I get an error message telling “Class Not found” while trying to register the listener. On windows phone I get nothing.

Following are the steps I used to create the application using the cordova CLI

Created a cordova application

Added platforms

Added the plugin (https://github.com/chariotsolutions/phonegap-nfc.git)

Added the following code replacing deviceready in index.js file 

    try {
        // Read NDEF formatted NFC Tags
        nfc.addNdefListener(
            function (nfcEvent) {
                var tag = nfcEvent.tag,
                    ndefMessage = tag.ndefMessage;

                // dump the raw json of the message
                // note: real code will need to decode
                // the payload from each record
                alert(JSON.stringify(ndefMessage));

                // assuming the first record in the message has 
                // a payload that can be converted to a string.
                alert(nfc.bytesToString(ndefMessage[0].payload).substring(3));
            },
            function () { // success callback
                alert("Waiting for NDEF tag");
            },
            function (error) { // error callback
                alert("Error adding NDEF listener " + JSON.stringify(error));
            }
        );
    } catch (ex) {
        alert(ex.message);
    }

    app.receivedEvent('deviceready');

}

Compressed the project folder and uploaded it to phone-gap-build.

Got the project build and deployed to android and windows 8.1 phones (nfc was enabled on both devices). Phone-gap build has used PhoneGap 3.7.0 to build the application

When I tried to execute the app, while trying to register the listener, Android complained ‘Class not found’. Windows phone did not throw any errors (at least nothing I could see), but was not recognizing any of the NFC cards presented to it. Under \nfcReaderB\platforms\android\src\com\chariotsolutions\nfc\plugin, I can see the required java source files for android platform

NFC functions on the phones are working fine

The code is available at https://github.com/cmeegamarachchi/nfc

And help on resolving this is much appreciated

don, 2015-06-19:
Class not found usually means that PhoneGap can't find the plugin.

I cloned your project and it ran on android with Cordova 5.0, so I'm thinking this is PhoneGap build problem. Or perhaps, just a version problem?

    $ git clone https://github.com/cmeegamarachchi/nfc.git
    $ cd nfc
    $ cordova run android --device

I'll see if I can get it to work with PhoneGap later

cmeegamarachchi, 2015-06-20:
Many thanks for the quick feedback Don. 

Please do let me know if you could get it to run with phone-gap

Chintana

don, 2015-10-24:
Your repo works with PhoneGap 5.3.1 on Android when building locally.

I'm guessing this is a PhoneGap Build problem. See if you can install the plugin from NPM by editing config.xml http://docs.build.phonegap.com/en_US/configuring_plugins.md.html

closed 2015-10-24 by don

don, 2015-10-30:
@cmeegamarachchi To get your repository to work with phonegap build, I added the gap namespace and the phonegap-nfc plugin to config.xml. 

    <gap:plugin name="phonegap-nfc" source="npm" />

See the fork of your project for more info See https://github.com/don/phonegap-nfc-issue-190

mountainclimbing, 2018-06-04:
Where exactly do you add this, "<gap:plugin name="phonegap-nfc" source="npm" />" in the config.xml file. Am sure it should be just at the bottom