chariotsolutions/phonegap-nfc #191: PhoneGap Build issue

ccchai, 2015-06-29:
For some reasons, I can't get 'git' to work in Cordova CLI, so I am creating my app using PhoneGap Build NFC plugin. It is working perfectly until I add another PhoneGap Build plugin, and my app fails to build for Android platform with the following error:

Error - Two of the plugins in your app are incompatible as they both add a file with the same name.

I suspect the problem is caused by this file:

 <source-file src="src/android/org/apache/cordova/api/Dummy.java" target-dir="src/org/apache/cordova/api"/>

is it possible to offer a PhoneGap Build plugin without this Dummy.java?

don, 2015-06-29:
Dummy.java is added for compatibility with old versions of Cordova when there was an API change. If you're using PhoneGap 5 or Cordova 5 you don't  need this file. Fork one of the plugins on github and remove Dummy.java from the source tree and plugin.xml. I'll remove this in a future release of the plugin.

Also see https://github.com/don/BluetoothSerial/issues/145

resonantbreath, 2015-09-11:
+1 the current repository of plugins will be disconnected October 15th 2015 and will be accessed with 3.0+ only fairly soon

resonantbreath, 2015-09-11:
You can use my fork, it removes the Dummy.java
`cordova plugin add https://github.com/resonantbreath/phonegap-nfc --save`

don, 2015-10-23:
Dummy.java was removed in version 0.6.4

closed 2015-10-23 by don