chariotsolutions/phonegap-nfc #197: App goes into background for a moment

m-menon, 2015-07-30:
hi,

Firstly thanks a lot for the useful plugin. I am a little new to java / android and facing a small problem with the plugin. My app runs the camera and when someone taps his NFC card on the phone, it just records the NFC card details. All that works perfectly fine except for one thing - when the NFC card is being read, the App goes into background momentarily - for something like 10 - 50 Millisecond ( I got this figure from the Pause / Resume events ). Now this causes the Camera to freeze / hang / crash. Any clues or leads why this could be happening? I have tried a zillion things I could - but it still hangs, crashes. Any help or suggestion would be great.

As for reading the NFC card I am just using this simple code
```
function nfcHandler (nfcEvent) {
    var tag = nfcEvent.tag;
    var ndefMessage = tag.ndefMessage;
    var payload = nfc.bytesToString(ndefMessage[0].payload);
    alert(payload);
};

nfc.addNdefListener (
    nfcHandler,
    function () {
        console.log("Success, listener added. You can now scan a tag.");
    },
    function (error) {
        alert("Adding the listener failed.");
    }
); 
```

And this for removing the listener when I am done

```
nfc.removeNdefListener(
    nfcHandler, // this must be the same as the function above
    function () {
        console.log("Success, the listener has been removed.");
    },
    function (error) {
        alert("Removing the listener failed");
    }
);
```

Thanks,
M&M

don, 2015-08-10:
Maybe you can make a call in your nfcHandler to put the camera in a state where it won't crash?

m-menon, 2015-08-10:
Hi,

Yup i tried that. I tried stopping the camera the moment the nfc is read. I have tried hiding the camera. I have tried hide + stop. It all leads to camera freeze and a crash. I have created a small app just to recreate the problem. Can i send you the apk or the eclipse project?

My surprise is that why does my app take a backseat momentarily. Is it that the built in android nfc service is taking control when the card is first detected and dispatching the event to the registered app or something 

Thanks a lot for your time. I have tried everything within my capacity but the solution still eludes me.

Thanks,
MnM

BoonChew, 2015-09-20:
Hi guys, 

i am new to Cordova plugin. I am developing an Android Apps with IBM Worklight. I just want to read the content of an NFC tag with my Android apps.  But i don't know how to start it. can somebody teach me in detail? 

your help is appreciated.

Thanks in advance.

don, 2015-09-21:
@BoonChew it's best practice to open a new issue for each topic, rather than start a new conversation on an existing issue.

Take a look at [GettingStarted](https://github.com/chariotsolutions/phonegap-nfc/blob/master/doc/GettingStartedCLI.md). The guide is a bit old, so if you run into a problem open a new issue. There are also [sample projects](https://github.com/chariotsolutions/phonegap-nfc#sample-projects) and a [Beginning NFC book](https://github.com/chariotsolutions/phonegap-nfc#book). The [project page](https://github.com/chariotsolutions/phonegap-nfc) has a lot of info to get you started.

don, 2015-11-17:
I think this is app specific rather than an issue with the plugin. Closing for now.

closed 2015-11-17 by don

andreujuanc, 2017-06-02:
@don This just happened to me 😮 
@m-menon did you found workaround for this?