chariotsolutions/phonegap-nfc #198: Turn off default / Android NFC sound

m-menon, 2015-08-03:
I am playing a sound in my app when I scan a NFC card using your plugin. But Android plays an annoying notification sound too when it scans a tag - and that is by design :-(

Is there any way that we could stop the annoying notification from Android? :-) I came across this and It's supported on API level >= 19. Any thoughts?

And also thanks a ton for the plugin. It has really been of immense help in my application. I believe it's guys like you who greatly contribute, enhance to the quality of software being developed

Here's some info on turn off the sound albeit it's for level 19 and above
with android-19 you can:

as described in: http://developer.android.com/reference/android/nfc/NfcAdapter.html#FLAG_READER_NO_PLATFORM_SOUNDS

By using NfcAdapter.enableReaderMode() and the flag FLAG_READER_NO_PLATFORM_SOUNDS instead of NfcAdapter.enableForegroundDispatch()

Thanks
M&M

don, 2015-10-26:
I looked into this a bit... [FLAG_READER_NO_PLATFORM_SOUNDS](http://developer.android.com/reference/android/nfc/NfcAdapter.html#FLAG_READER_NO_PLATFORM_SOUNDS) works with [enableReaderMode](http://developer.android.com/reference/android/nfc/NfcAdapter.html#enableReaderMode(android.app.Activity, android.nfc.NfcAdapter.ReaderCallback, int, android.os.Bundle)) which is an alternate way of doing foreground dispatch for NFC. The new API is simpler than [enableForegroundDispatch](http://developer.android.com/reference/android/nfc/NfcAdapter.html#enableForegroundDispatch(android.app.Activity, android.app.PendingIntent, android.content.IntentFilter[], java.lang.String[][])) than since rather than Intents, the Tag is sent to a [NfcAdapter.ReaderCallback](http://developer.android.com/reference/android/nfc/NfcAdapter.ReaderCallback.html). Processing with a callback is different from the current Intent processing that that plugin uses. It might be possible to work around this. 

The deal killer is that using `NfcAdapter.enableReaderMode` completely disables the NFC peer-to-peer functionality.

jlkalberer, 2015-11-07:
I would be OK with p2p being disabled :)

jezdez, 2016-01-09:
@don I'd also be interested in this and wondered if the simpler reader mode could be supported optionally for use cases in which only reading nfc tags is needed and no p2p mode or card emulation?

jezdez, 2016-01-09:
FYI, I opened a PR with what I think this could look like, no idea if that makes sense, thought I'd give it a try anyways. Any takers to review it?

obiwarn, 2016-01-14:
+1 (but too stupid to review by myself)

don, 2018-06-01:
I have a working version of enableReaderMode on a [branch](https://github.com/chariotsolutions/phonegap-nfc/tree/reader-mode). I need to test some more to see how it effects the other NFC functions. Currently, it looks like foreground dispatching is disabled when reader mode is enabled.

Install from the branch

    cordova plugin add https://github.com/chariotsolutions/phonegap-nfc.git#reader-mode

To scan NDEF tags with no system sounds try

    nfc.readerMode(
        nfc.FLAG_READER_NFC_A | nfc.FLAG_READER_NO_PLATFORM_SOUNDS, 
        tag => console.log(JSON.stringify(tag))
    )

Note that the tag object is passed directly to the successCallback of `nfc.readerMode`. There's no separate function that listens for events. See the [Android docs](https://developer.android.com/reference/android/nfc/NfcAdapter.html#enableReaderMode(android.app.Activity,%20android.nfc.NfcAdapter.ReaderCallback,%20int,%20android.os.Bundle)) for more details about the flags.

m-menon, 2018-06-01:
That's cool! Thanks

closed 2018-06-06 by don

don, 2018-06-06:
Reader mode is published in [version 1.0.2]( https://www.npmjs.com/package/phonegap-nfc/v/1.0.2)

PR #326 (2018-06-06): Reader mode
Fixes #198