chariotsolutions/phonegap-nfc #199: NTAG216 Set Lock Bit

JohnMcLear, 2015-08-03:
NTAG216 introduced the ability to set lock bits.

"Configurable password protection with optional limit of unsuccessful attempts" - Source: http://www.nxp.com/documents/data_sheet/NTAG213_215_216.pdf

Amal's Dangerous Things Android App does this I assume by directly writing bits on the tag, see https://dangerousthings.com/shop/xnt-ntag216-2x12mm-glass-tag/

It would be nice to have this ability baked into the phonegap-nfc app but I would expect this would be an android only implementation as I doubt other providers expose the required methods.

don, 2015-10-24:
Might be related to #176

chelinho139, 2019-03-26:
Hey, any updates or PR's on password protected ntag216?