chariotsolutions/phonegap-nfc #200: Question: Is is possible to disable Beam (share)

tsukasa1989, 2015-08-06:
Hello,

Is it possible with the current version to disable NFC Beam? Because for some screens of my application I do not want Android to "zoom" out so the Android Beam takes over. 

I know you can do this within Android:

    NfcAdapter adapter = NfcAdapter.getDefaultAdapter(this); 
    adapter.setNdefPushMessage(null, this, this);

Is there any way to simulate that action? To pass null into the first param.

Thanks

tsukasa1989, 2015-08-06:
nfc.unshare(); already calls this method, so on startup I just call nfc.unshare();

closed 2015-08-06 by tsukasa1989

don, 2015-08-10:
I think that Android defaults to beaming an AAR record for your app if you don't define one. Calling `nfc.unshare()` will work. I think there's also a way to disable this in the AndroidManifest.xml too, but I can't find a link right now.