chariotsolutions/phonegap-nfc #201: WINDOWS PHONE PLUGIN 8 DOES NOT WORK

sainzrow, 2015-08-06:
ON WINDOWS PHONE 8 PLUGIN DOES NOT WORK

sainzrow, 2015-08-07:
Fix this fu***** plugin

allannaranjo, 2015-08-07:
Wow, are you for real?!!!!
sainzrow, if it does not work for you then MAKE YOUR OWN PLUGIN, github is a community of developers helping each other, not demanding like you are.
I'm glad it does not work for you!, and hope you are in a hurry!, sad little person!

sainzrow, 2015-08-07:
why write on suported platforms like wp8 if the plugin does not work, do not post for wp8 if people are not abble to make a plugin for specified device

allannaranjo, 2015-08-07:
Make it work in windows phones, make a pull request and help instead of making this ("Fix this fu***** plugin") type of comments, and if the plugin does not work, add some logs, exceptions etc to the issue.

sainzrow, 2015-08-07:
testing on device Lumia 820 'TaskHost.exe' (CoreCLR: Silverlight AppDomain): caricamento di 'C:\windows\system32\System.Core.ni.dll' completato. Caricamento dei simboli ignorato. Il modulo è ottimizzato e l'opzione del debugger 'Solo codice utente' è abilitata.
Adding feature.value=NfcPlugin
CordovaBrowser_LoadCompleted
Apache Cordova native platform version 3.8.0 is starting
Received Event: deviceready
'TaskHost.exe' (CoreCLR: Silverlight AppDomain): caricamento di 'C:\windows\system32\System.ServiceModel.Web.ni.dll' completato. Caricamento dei simboli ignorato. Il modulo è ottimizzato e l'opzione del debugger 'Solo codice utente' è abilitata.
'TaskHost.exe' (CoreCLR: Silverlight AppDomain): caricamento di 'C:\windows\system32\System.Runtime.Serialization.ni.dll' completato. Caricamento dei simboli ignorato. Il modulo è ottimizzato e l'opzione del debugger 'Solo codice utente' è abilitata.
'TaskHost.exe' (CoreCLR: Silverlight AppDomain): caricamento di 'C:\windows\system32\en-US\mscorlib.debug.resources.dll' completato. Modulo compilato senza simboli.
Nfc Plugin
Registering for NDEF

sainzrow, 2015-08-07:
and on scan nothing happends :(
trying with other nfc reader app downloaded on windows store, nfc works

closed 2015-08-07 by sainzrow

don, 2015-08-10:
This plugin uses Silverlight and C# for the WP8. It might need some work in Visual Studio too get it to do the right thing with the latest Cordova.