chariotsolutions/phonegap-nfc #202: Cannot Make it Run on a BlackBerry z30 SmartPhone

LaermTot, 2015-08-20:
At the Moment I olny want to create a Simple Cordova-App for a Blackberry Z30

therefore I creates a new a new App  using

> cordova create meh com.gorilla.meh meh
> cordova platform add blackberry10 
> cordova plugin add com.blackberry.invoke
> cordova plugin add https://github.com/chariotsolutions/phonegap-nfc.git

Them I modified config.xml 

<?xml version='1.0' encoding='utf-8'?>
 .. the generated Code comes here....
  \<rim:invoke-target id="com.gorilla.meh.target"\>
    \<type\>APPLICATION\</type\>
    \<filter\>
        \<action\>bb.action.OPEN\</action\>
        \<mime-type\>application/vnd.rim.nfc.ndef\</mime-type\>
         \<property var="uris" value="ndef://0,ndef://1,ndef://2,ndef://3,ndef://4" /\>
    </filter>
</...>

Then I modified the www/js/index.js File 
by Adding a few Lines to onDeviceReady: function() {

onDeviceReady: function() {
        app.receivedEvent('deviceready');								 
		nfc.addNdefListener (
			function (nfcEvent) {					
				var tag = nfcEvent.tag,
				ndefMessage = tag.ndefMessage;			
				alert(JSON.stringify(ndefMessage));            
				alert(nfc.bytesToString(ndefMessage[0].payload).substring(3));
			}, 
			function () { // success callback
				alert("Waiting for NDEF tag");
			},
			function (error) { // error callback
				alert("Error adding NDEF listener " + JSON.stringify(error));
			}
    );

But this Does not work .. I get no Results .. and it seems that the application of the Event-Listener nfc.addNdefListener seems to fail .. perhapes it's undefined .. 

Any Ideas what I am Missing ??

I Tried to compile the https://github.com/don/phonegap-nfc-reader Demo-Source but I wonÄt work .. It complains about  ""C:\PHONEG~1\PLATFO~1\BLACKB~1\cordova\init"" not being visible ,, seems to be a strange DOS Problem .. 

Any Help is Welcome =)

Volker

don, 2015-08-20:
I haven't looked at BB in a long time. It's possible something changed with the SDK?

Does the tag you're reading work with other apps?
Can you connect the remote debugger and see if there are runtime issues with your code?
I'll see if I can track down a BB10 device and get something to run.

LaermTot, 2015-08-21:
Well I have to admit that I never managed to mage to make the remote Debugger work .. I can connect to the provided URL but IN the End I See nothing that gives me a hint .. 

Any Ideas concerning the DOS-Naming of the Path ?

LaermTot, 2015-08-31:
Any News / Ideas ??

Greeting Volker

don, 2015-09-11:
I found my Z10, but it won't power up. Letting it charge for a while, hoping it comes back to life.

don, 2015-10-26:
I got this working on BB10 again. You'll need version 0.6.5 of the plugin.

Hardware & Software
 * OSX 10.10.5 
 * BB10 WebWorks SDK 2.2.0.15
 * Cordova 5.5.3
 * PhoneGap NFC 0.6.5
 * BlackBerry Z10 running 10.3.1.156

Create the project and install the plugins

    cordova create nfc com.example.nfc HelloNFC
    cd nfc
    cordova platform add blackberry10
    cordova plugin add phonegap-nfc
    cordova plugin add cordova-plugin-bb-invoke

edit `www/js/index.js` and replace `onDeviceReady` with

    onDeviceReady: function() {
        app.receivedEvent('deviceready');

        // Read NDEF formatted NFC Tags
        nfc.addNdefListener (
            function (nfcEvent) {
                var tag = nfcEvent.tag,
                    ndefMessage = tag.ndefMessage;

                // dump the raw json of the message
                // note: real code will need to decode
                // the payload from each record
                alert(JSON.stringify(ndefMessage));

                // assuming the first record in the message has
                // a payload that can be converted to a string.
                alert(nfc.bytesToString(ndefMessage[0].payload).substring(3));
            },
            function () { // success callback
                alert("Waiting for NDEF tag");
            },
            function (error) { // error callback
                alert("Error adding NDEF listener " + JSON.stringify(error));
            }
        );
    },

Edit `config.xml` and `<rim:invoke-target>` just before `</widget>`. (See docs for more info about [invoke-target](https://github.com/chariotsolutions/phonegap-nfc#blackberry-10-invoke-target
).)

    <rim:invoke-target id="your.unique.id.here">
        <type>APPLICATION</type>
        <filter>
            <action>bb.action.OPEN</action>
            <mime-type>application/vnd.rim.nfc.ndef</mime-type>
            
            <property var="uris" value="ndef://0,ndef://1,ndef://2,ndef://3,ndef://4" />
        </filter>
    </rim:invoke-target>

Compile, install, and run

    cordova run

[image]

don, 2015-10-26:
@LaermTot if this doesn't fix your problem, add a comment and I'll re-open the issue

closed 2015-10-26 by don

LaermTot, 2015-10-27:
You are my HERO =) ...... thanx it works now !

LaermTot, 2015-11-13:
Just another stupid Question .. how do I initilize the Event-Listener using "jquery mobile" ?? The was deswcribes above, onDeviceReady: function() { is never reached .. any Ideas ??