chariotsolutions/phonegap-nfc #204: Add support for non-NDEF tags on Windows Phone

aleksandr-m, 2015-09-10:
Seems like WP8.1 supports some non-NDEF tags.

From [What's New in Windows Phone 8.1](https://msdn.microsoft.com/library/windows/apps/dn632424.aspx):
>In Windows Phone 8.1 we also expand the supported NFC tag scenarios so you can make your tags read-only, write to unformatted tags, and communicate with other contactless cards. Support for these scenarios requires NFC hardware that's using a PN547 chip.

Also Stack Overflow question: [Does WP8.1 Support Mifare Classic read?](http://stackoverflow.com/q/23626495/1700321)

don, 2015-10-23:
If the newer WP devices have the NXP chipset, they should be to read and write NDEF messages to Mifare Classic tags, even though MifareClassic isn't an official NFC Forum tag type.

The new APIs should make it possible to implement things like [nfc.makeReadOnly](https://github.com/chariotsolutions/phonegap-nfc#nfcmakereadonly). Android handle format internally, but this could be exposed as a new method.

This plugin supports NDEF since it works well across tag-types and operating systems. At this time don't plan on adding non-NDEF features unless clients need them.  If support for other tags can be added cleanly and across platforms, I can accept pull requests.

don, 2015-11-16:
@JohnMcLear your NFC Ring issue clarified this a bit. Writing NDEF messages to unformatted cards is a good idea and should be supported.

I tried to implement [nfc.addTagDiscoveredListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddtagdiscoveredlistener) for WP, but I don't see a way to get notified when a non-NDEF tag is scanned.

From reading the docs it seemed like proximityDevice.subscribeForMessage("WriteableTag", handler); does this, but it only works for NDEF tags. Maybe there's something that can be done with PeerFinder and triggeredconnectionstatechanged, but I couldn't get it to work.

**There is a work around** (even though I don't like it.) On Windows Phone, you can call `nfc.write(ndefMessage)` at any time, not just when a tag is in range. This is very different than Android. WP queues the message and writes to the tag when a tag enters the field. For my tests, WP formatted empty Mifare Ultralight tags and some Mifare Classic tags.

FWIW, the WP format of the tags broke something on the tag. They work fine as NDEF tags but can't be reset anymore.

don, 2015-11-18:
[nfc.addTagDiscoveredListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddtagdiscoveredlistener) is now supported on Windows. If you have the right NXP chipset in your phone (e.g. Lumia 640), you can use [nfc.addTagDiscoveredListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddtagdiscoveredlistener) to be notified when a tag appears to the phone. The [tag event](https://github.com/chariotsolutions/phonegap-nfc#non-ndef-tag-scanned-with-addtagdiscoveredlistener-on-windows) will not contain any data. Use [nfc.write](https://github.com/chariotsolutions/phonegap-nfc#nfcwrite) to write a new NDEF message to the tag. Windows will automatically format the tag before writing the NDEF message.

/cc @JohnMcLear

closed 2015-11-18 by don

JohnMcLear, 2015-11-18:
Awesome!!  I will try to get an update out with this implemented, at the moment I do a simple if statement for the device OS so I can probably drop the IF entirely..

JohnMcLear, 2015-11-18:
Thanks btw :)

don, 2015-11-18:
@JohnMcLear You're welcome. BTW, this will be in 0.6.6 which will be in NPM in a few minutes.

JohnMcLear, 2015-11-20:
Needs testing https://github.com/mclear/NFC_Ring_Control/pull/88

JohnMcLear, 2015-11-25:
``    nfc.addTagDiscoveredListener(function (nfcEvent) {`` not working on  whatever thing I have..  Will continue to investigate..

don, 2015-11-25:
@JohnMcLear I don't think the 820 has the NXP547 chipset so I'm not sure if it will be able to see the non-ndef tags.

It sounds like you have another issue with the upgrade from 0.4.7 to 0.6.6. Remove and add should get the latest plugin from NPM. If you're seeing the old version something is cached. 

Upgrading shouldn't break the app. Did you get a newer version of Cordova too? Open a new issue for this, I'd like to try and duplicate. Put links to the github repo and versions.

JohnMcLear, 2015-11-25:
https://github.com/mclear/NFC_Ring_Control uses 0.4.7
upgrading  to
0.6.6 @ https://github.com/mclear/NFC_Ring_Control/tree/winmo-non-ndef breaks NFC

STR

checkout master branch, run, test, ensure NFC works.
``cordova remove phonegap-nfc`` then ``cordova add phonegap-nfc`` should get 0.6.6 but doesn't so remove then add from git.

Try to run with 0.6.6 and notice all NFC functionality is broken in WinMo