chariotsolutions/phonegap-nfc #206: The tag ID

clarklight, 2015-10-06:
I have a quick question about the tag ID. 

1.) Is it unique? 
2.) Can a user override it? 

I just want to know if i can rely on it being secure. Or if i would build a unique set of ID myself then lock the card up??

don, 2015-10-08:
It depends on the tag type. Mifare Classic 4 byte ids are reused. I think most 7 byte ids on other tags should be unique, but you'd need to check with the tag manufacturer to be sure.

You're not supposed to be able to change the tag id. I've heard of non-conformant tags that let you change the id, but I've never used one.

allannaranjo, 2015-10-08:
Even you cannot prevent users from cloning your tag you can store the Id
and make sure you store the tag id associated with the content.
To encrypt I use an sha256 library.
Allan

clarklight, 2015-10-08:
@allannaranjo  but if someone use a card emulator and clone the whole thing, and when the server reads it and decrypt it, the replica card would be able to cause action to the db tho. but thanks for the advice anyway. 

On NXP documentation they said they have the "anti-cloning" mechanism but i am not sure if it is as tough as it says....

closed 2015-10-23 by don