chariotsolutions/phonegap-nfc #207: Suppress OS messages in Windows Phone

JohnMcLear, 2015-10-12:
This can be achieved in 8.1

http://stackoverflow.com/questions/18964181/disable-default-windows-phone-nfc-reader

don, 2015-10-23:
@JohnMcLear The code [win8](https://github.com/chariotsolutions/phonegap-nfc/blob/master/src/windows-phone-8/NfcPlugin.cs#L53) and [windows](https://github.com/chariotsolutions/phonegap-nfc/blob/master/src/windows/nfc-plugin.js#L195) is registering for NDEF subscriptions like the [selected answer](http://stackoverflow.com/a/27976390/1416463) on StackOverflow says. If you're still seeing this OS messages with Windows 8.1 there might need some other updates required?

Unfortunately I can't test this. My Lumia 825 refuses to update past WP8.0 and my newer WP devices don't have NFC.

JohnMcLear, 2015-10-25:
Tested confirmed working on 8.1!   Awesome thanks!

closed 2015-10-25 by JohnMcLear