chariotsolutions/phonegap-nfc #213: Fail gracely on unsupported platforms

mobidev111, 2015-11-15:
On windows 8.1 (not windows phone) the plugin fails with the following message:
```
   adding proxy for NfcPlugin
   WARNING: proximity device is null
   Registering for NDEF
   TypeError: Unable to get property 'subscribeForMessage' of undefined or null reference
   adding NFC Listener failed: TypeError: Unable to get property 'subscribeForMessage' of undefined or null reference
   Error in Error callbackId: NfcPlugin816380662 : adding NFC Listener failed: TypeError: Unable to get property 'subscribeForMessage' of undefined or null reference
   Exception calling native with command :: NfcPlugin :: registerNdef ::exception=adding NFC Listener failed: TypeError: Unable to get property 'subscribeForMessage' of undefined or null reference
   Initialized the NfcPlugin
```

-> better fail by testing for supported platforms

JohnMcLear, 2015-11-16:
8.1 is working for me, can you provide your source code so we can replicate please?

don, 2015-11-18:
Fixed in 0.6.6

closed 2015-11-18 by don