chariotsolutions/phonegap-nfc #220: NFC Service detects  tag before addNdefFormatableListener

cappelaere, 2015-12-17:
I am trying to detect an NDEF tag and write to it.
I use nfc.addNdefFormatableListener(...)
It does go into listening mode for NDEF Formatable tags...
Problem is that when I tap a tag, the NFC service detects it and brings up a popup before my listener detects it. 
It says: 
New tag collected
Empty tag

 How can I turn that off?
I am running Android 6.0. on Nexus 6P
Sooo close....Thanks,
Pat.

don, 2015-12-17:
It sounds like you have an error in your app. If the listener gets added, the OS should not see that tag, since the app uses foreground dispatching.

Check for errors in `adb logcat` and with chrome://inspect.

cappelaere, 2015-12-17:
Right.  I wish I could find this... grrr!
Here is the controller code snippet... May be there is something I am completely overlooking.
```javascript
.controller('NFCCtrl', function($scope) {
	console.log("NFCCtrl...");
	$scope.settings = {
		url: 	"http://localhost/test/Nk@l2LI6S7x-V1MlhLL6Smg",
		msg: 	undefined
	};
	
	nfc.addNdefFormatableListener(
	    function (nfcEvent) {
			$scope.settings.msg 	= "addNdefFormatableListener..."
	
			var message = [
				ndef.uriRecord( $scope.settings.url)
			];

			nfc.write(
				message, 
				function() {
					$scope.settings.msg 	= "Wrote url to tag:"+	$scope.settings.url
					
				},
				function(error) {
					$scope.settings.msg 	= "Err:"+JSON.stringify(error)	
				}
			);
		}, 
		function () {
			$scope.settings.msg 	= "Listening for NDEF Formatable Tags";
		},
		function (error) {
			$scope.settings.msg 	= "Failed to add listener for NDEF Formatable Tags";  
		}
	);
})
```
I do not see any errors.
I do see the Listening for NDEF... message
Then NFC Service takes over.
Sorry to bother you for this :(
Pat.

don, 2015-12-17:
I see what's happening now.

Since you're seeing "New tag collected, Empty tag" that means it's an NDEF tag with an empty record. The NDEF Formatable Listener won't find this tag, since it's already formatted. You need a Tag Discovered Listener or NDEF Listener. (These are Android's rules, not mine.) You can add multiple listeners, Android will call the most specific one for a tag.

I don't have any unformatted tags with me. I tried using format.js from [mifare-classic-js](https://www.npmjs.com/package/mifare-classic) to erase tags. (If you just want to unformat MifareClassic tags, you can use libfreefare's mifare-classic-format command)

Android 6.0 finds the tag but writing fails with Nexus 5X. The same code works fine on Samsung SIII (Android 4.3) and Samsung Galaxy (Android 4.2.2). I don't and Android 5 phone with me to test.

`format.js` creates an NDEF formatted tag, without any NDEF message. (Technically I think this breaks the spec.) It would be better to have an completely reset tag using the non-ndef keys for each sector.

I have an [Arduino sketch](https://github.com/don/NDEF/blob/1eb9a098c1e2d5ba7d3f16636f9b611c48c3704c/examples/CleanTag/CleanTag.ino) to Clean Mifare Classic and Mifare Ultralight. I'll have to try that later.

Here's the test code I was using.

    cordova create issue220 com.example.issue220 issue220
    cd issue220
    cordova plugin add phonegap-nfc
    cordova platform add android

Replace www/js/index.js with 

    var app = {
        initialize: function() {
            this.bindEvents();
        },
        bindEvents: function() {
            document.addEventListener('deviceready', this.onDeviceReady, false);
        },
        onDeviceReady: function() {
            app.receivedEvent('deviceready');
            var success = function() {
                console.log("Listening for formatable NFC tags");
            };
            var failure = function(reason) {
                console.log("Error adding NFC listener " + reason);
            };

            nfc.addNdefFormatableListener(app.onNfc, success, failure);
        },
        onNfc: function(nfcEvent) {

            var success = function() {
                alert("Wrote data to tag");
            };
            var failure = function(reason) {
                alert("NFC write failed " + reason);
            };

            var message = [
                ndef.uriRecord("http://www.amazon.com/dp/1457187094")
            ];

            nfc.write(message, success, failure);

        },
        // Update DOM on a Received Event
        receivedEvent: function(id) {
            var parentElement = document.getElementById(id);
            var listeningElement = parentElement.querySelector('.listening');
            var receivedElement = parentElement.querySelector('.received');

            listeningElement.setAttribute('style', 'display:none;');
            receivedElement.setAttribute('style', 'display:block;');

            console.log('Received Event: ' + id);
        }

    };

    app.initialize();

FWIW In my Angular example, I put the NFC code in a service https://github.com/don/ionic-nfc-reader.

cappelaere, 2015-12-17:
Don,

I think that you are right on all counts.
The NDEF Formatable Listener won’t find this tag (That’s why the NFC Service found it behind the scene… which surprised me)
The NDEF Listener find the tag but writing is not happening on Android 6.0 Nexus 6P.  I do not get an error either.
The tag is NTAG 213.
More to follow… Really appreciate your support.
Pat.

cappelaere, 2015-12-17:
Don,

I have to admit that I am very puzzled now.  I updated to 6.0.1 and shuffled some code around so I could remove the listeners.
Now it is working.  I can write data to the tag.  I am really not sure exactly what change caused this.
I am very sorry.  I am trying to understand this a little better and see if I can provide more info.

Pat.

don, 2015-12-17:
If you share your code somewhere I can take a look and try it on my devices.

cappelaere, 2015-12-18:
Don,

Here is a link to my repo: https://github.com/vightel/coinRwApp <https://github.com/vightel/coinRwApp>
Everything seems to work fine on Android 6.0.1 and NEXUS 6P
This is my only requirement for this project.
Thanks for your great work and amazing library.
I owe you one.

Pat Cappelaere
pat@cappelaere.com

Cell: 410 340 4868
Skype: patrice_cappelaere
gmail: cappelaere@gmail.com

don, 2018-04-18:
Closing old issue

closed 2018-04-18 by don