chariotsolutions/phonegap-nfc #221: Windows 10 Mobile

ruisilva450, 2015-12-21:
Hi,
I tried to use your plugin on a Lumia 920 and 950XL and both returned NO_NFC_OR_NFC_DISABLED everytime on ```nfc.addNdefListener(callback, [onSuccess], [onFailure]);```

I have NFC enabled on both devices

Thank you

don, 2015-12-21:
I haven't run with with Windows 10 yet, just Windows 8.1 on my Lumia 640.  Maybe the API or permissions changed for Windows 10? Are you seeing any error messages in Visual Studio?

ruisilva450, 2015-12-22:
No. And I'm seeing logs that the plugin is successfully recognizing NFC tags.

don, 2015-12-22:
It looks like this is a permission issue.

I had a Cordova app on the phone before upgrading the phone to Windows 10 and it ran fine after upgrading. It was able to read NFC tags.

I recreated the app and deployed from Visual Studio 2015 onto the Windows 10 phone and I got a permission error.

[image]

To recreate the app I was using for debugging.

    cordova create nfc com.example.nfc NFC
    cd nfc
    cordova platform add windows
    cordova plugin add phonegap-nfc

edit www/js/index.js and replace `onDeviceReady` with

    onDeviceReady: function() {
        app.receivedEvent('deviceready');

        // Read NDEF formatted NFC Tags
        nfc.addNdefListener (
            function (nfcEvent) {
                var tag = nfcEvent.tag,
                    ndefMessage = tag.ndefMessage;

                // dump the raw json of the message
                // note: real code will need to decode
                // the payload from each record
                alert(JSON.stringify(ndefMessage));

                // assuming the first record in the message has
                // a payload that can be converted to a string.
                alert(nfc.bytesToString(ndefMessage[0].payload).substring(3));
            },
            function () { // success callback
                alert("Waiting for NDEF tag");
            },
            function (error) { // error callback
                alert("Error adding NDEF listener " + JSON.stringify(error));
            }
        );
    },

Open Visual Studio, switch the project to CordovaApp.Windows10 (Universal Windows) and deploy to the phone.

If you can figure out the correct permissions in Visual Studio, I can update the manifest to do this automatically.

ruisilva450, 2015-12-23:
I saw you code and it appears right. I changed plugin.xml to add this:

```C#
<config-file target="package.appxmanifest" parent="/Package/Capabilities">
    <DeviceCapability Name="proximity" />
</config-file>
```

Notice that since we are dealing with Universal Apps it makes sense that the target is package.appxmanifest and not "package.xxx.appxmanifest".

Still this doesn't work and I'm out of ideas.

don, 2015-12-28:
@ruisilva450 I think the trick is to get the generated Windows project to work, then copy those settings back to into cordova.xml.

tervoju, 2016-03-07:
any news on this? trying this in Samsung/android - working, Lumia 950 - not working.

lamuertepeluda, 2016-05-14:
+1 on this issue. I had the same error as @don and I had to add proximity capability manually for win 10 universal app in order to avoid that runtime error. I think you should just add one more line in the plugin.xml. I'll check how I did for another plugin and let u kno

lamuertepeluda, 2016-05-15:
This is working in my plugin.xml - taken from a different plugin not this one - to support windows phone 8.1, windows 8.1 and 10.

```xml
 <platform name="windows">
	<config-file target="package.appxmanifest" parent="/Package/Capabilities" versions="8.1.0">
		<m2:DeviceCapability Name="bluetooth.rfcomm">
			<m2:Device Id="any">
				<m2:Function Type="serviceId:00001101-0000-1000-8000-00805F9B34FB"/>
			</m2:Device>
		</m2:DeviceCapability>
	</config-file>
	<config-file target="package.appxmanifest" parent="/Package/Capabilities" versions=">8.1.0">
		<DeviceCapability Name="bluetooth.rfcomm">
			<Device Id="any">
				<Function Type="serviceId:00001101-0000-1000-8000-00805F9B34FB"/>
			</Device>
		</DeviceCapability>
	</config-file>
</platform>
```

I know the config-file items seem identical but without declaring it explicitly I was not getting win 10 manifest to be updated... I think it is because of Cordova Windows platform.  
I guess you should use this pattern instead to get the manifest updated for Windows 10 as well. I will try to make a PR as soon as possible.

rancano, 2016-05-18:
I have the same issue, with nfc.addNdefListener. When i use nfc.addTagDiscoveredListener it detects the tag, but i don't have any information writen on it.

mobidev111, 2016-06-29:
Which 'DeviceCapability' settings need to be added? bluetooth.rfcomm or proximity or both?

Goal: read NFC tags in windows 10 universal apps - especially on windows 10 desktop with nfc reader

@lamuertepeluda @ruisilva450 What is needed for windows 10 desktop - Any updates or further findings? What else needs to be done?

lamuertepeluda, 2016-06-30:
Hi @mobidev111 , I'm sorry I didn't find much time to work on this plugin yet. Anyway I think you should only add `proximity` DeviceCapability, because the other one is needed for bluetooth rfcomm communication which are out of this plugin scope.

See [this link](https://msdn.microsoft.com/en-US/library/windows/apps/br211430.aspx?f=255&MSPPError=-2147217396) and [this other link](https://msdn.microsoft.com/en-us/windows/uwp/packaging/app-capability-declarations) for more info about capabilities.

Try if this work (I don't have access to a windows machine a.t.m. so I can't try it myself) by adding it to the ```plugin.xml``` file.

```xml
<platform name="windows">
    <config-file target="package.appxmanifest" parent="/Package/Capabilities" versions="8.1.0">
        <m2:DeviceCapability Name="proximity" />
    </config-file>
    <config-file target="package.appxmanifest" parent="/Package/Capabilities" versions=">8.1.0">
        <DeviceCapability Name="proximity" />
    </config-file>
</platform>
```

If the section for version=8.1.0 causes problems, remove it and leave the part with version>=8.1.0 which is for Windows 10.

mobidev111, 2016-06-30:
thx. this works for windows 10 mobile - without this I got the "Access Denied" message as described above (https://github.com/chariotsolutions/phonegap-nfc/issues/221#issuecomment-166757816)

Will check for windows 10 desktop.

@don can you add this to the plugin? it definitely gets people past the "Access Denied" message on windows 10 mobile.

jwillmer, 2016-07-27:
FYI: I did not encounter any `Access denied` issue as I compiled my project as a `Windows Phone (Universal)` app on a Lumia 950.

lamuertepeluda, 2016-07-27:
@jwillmer This issue occurs at runtime, not compile time. Tried on Nokia Lumia 1520.

jwillmer, 2016-07-27:
@lamuertepeluda also tested at runtime. Works as intended :+1:

andreujuanc, 2016-12-28:
Guys, this happened to me just today.
Quick fix was to addNdefListener after plugin is initialized.

I found this after I tried all described on this issue, and still didnt work.

The weird thing was also happened the same as @ruisilva450 , he said:

> No. And I'm seeing logs that the plugin is successfully recognizing NFC tags

So I was pretty sure NFC was enabled and working as expecting. But some how i just went and tried to see if i was getting subscribed too early (even if i do it ondeviceready).

First, I setted a breakpoint on phonegap-nfc.js line 19: `console.log("Initialized the NfcPlugin");`
Second, breakpoint at the line where I subscribe with addNdefListener.

Got the addNdefListener breakpoint stop first.

Settted a timer to wait a bit until plugin is initialized, and worked fine.

Should we create an event, so users can subscribe after plugin is initialzied?

Happy Holidays!

don, 2017-01-07:
#227 might fix this

don, 2018-04-18:
Closing old issue. RIP Windows Mobile. Maybe #265 handles for Windows10?

closed 2018-04-18 by don

andreujuanc, 2018-04-18:
@don  Miss Windows Mobile so much. RIP. 🍺