chariotsolutions/phonegap-nfc #223: Error registering listener  /   Failed to initialize the NfcPlugin Class not found

Fabio-sm, 2016-01-11:
Hi,
i'am trying to make a simple app with Cordova CLI to detect an NFC tag.
Those are the steps I've followed to create the app:
First I create the folder and the platform with Cordova CLI.
<pre>
<code>
cordova create nfc-app com.example.AppNfc AppNfc
cd nfc-app
cordova plugin add phonegap-nfc
cordova platform add android
</code>
</pre>

Then I used the comand: <code> cordova build android </code>.
Finally I used the comand: <code>phonegap serve</code> for testing the code on my smartphone.
But in the cmd I recived this error:
[image]

I post also my code:
<h5>Index.html</h5>
[image]

<h5>Index.js</h5>
<pre>
<code>
var app = {
    // Application Constructor
    initialize: function() {
        this.bindEvents();
    },
    // Bind Event Listeners
    //
    // Bind any events that are required on startup. Common events are:
    // 'load', 'deviceready', 'offline', and 'online'.
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
    },
    // deviceready Event Handler
    //
    // The scope of 'this' is the event. In order to call the 'receivedEvent'
    // function, we must explicitly call 'app.receivedEvent(...);'
    onDeviceReady: function() {
        app.receivedEvent('deviceready');
		
		nfc.addNdefListener(
			function(){
				//console.log(nfc.bytesToString(nfcEvent.tag.ndefMessage[0].payload));
				console.log("Tag");
			},
			function(){
				console.log("listener registered");
			},
			function(e){
				console.error("Error registering listener: " + e);
			}
		);
		
		nfc.addTagDiscoveredListener(
			function(){
				console.log("Tag");
			},
			function(){
				console.log("listener registered");
			},
			function(e){
				console.error("Error registering listener: " + e);
			}
		);
		
		nfc.addMimeTypeListener(
			"text/pg", 
			function(){
				console.log("Tag");
			},
			function(){
				console.log("listener registered");
			},
			function(e){
				console.error("Error registering listener: " + e);
			}
		);
		
		nfc.addNdefFormatableListener(
			function(){
				console.log("Tag");
			},
			function(){
				console.log("listener registered");
			},
			function(e){
				console.error("Error registering listener: " + e);
			}
		);
    },
    // Update DOM on a Received Event
    receivedEvent: function(id) {
        var parentElement = document.getElementById(id);
        var listeningElement = parentElement.querySelector('.listening');
        var receivedElement = parentElement.querySelector('.received');

        listeningElement.setAttribute('style', 'display:none;');
        receivedElement.setAttribute('style', 'display:block;');

        console.log('Received Event: ' + id);
    }
};

app.initialize();
</code>
</pre>

<h5>config.xml</h5>
[image]

I don't know what to do. I've read a lot of post and other issues but no one work for me.
Please help me, 
Thanks.

don, 2016-01-11:
"NfcPlugin Class not found" means that plugin is not installed. This is because you're "installing" on your phone with `phonegap serve`. In order for your app to work with `phonegap serve` you can only use the [plugins included with phonegap developer app](https://github.com/phonegap/phonegap-app-developer/blob/master/config.xml#L22-L45).

Try `cordova run android --device` to build the code and deploy to your phone.

Fabio-sm, 2016-01-11:
Thanks for your help. Now it's working.
I also try to use the APK generated by the instruction <pre>cordova build android</pre> and It function correctly.
Thank you so much!!!

closed 2016-01-12 by Fabio-sm