chariotsolutions/phonegap-nfc #228: NFC readonly Throws java.io.exception

abiodun0, 2016-03-06:
When I try to make a tag read-only using nfc.makeReadOnly([success], [failure]) on my command it shows java.io.exception and the tag can still be written to.

Anybody else have this problem?

don, 2016-03-07:
Can you run `adb logcat` and copy and paste the details of the exception?

taglme, 2016-05-11:
I get the same problem.
There is adb logcat log
`
E/AndroidRuntime( 9285): FATAL EXCEPTION: pool-1-thread-8
E/AndroidRuntime( 9285): Process: com.ionicframework.nfclabel717026, PID: 9285
E/AndroidRuntime( 9285): java.lang.IllegalStateException: Close other technology first!
E/AndroidRuntime( 9285): 	at android.nfc.Tag.setConnectedTechnology(Tag.java:458)
E/AndroidRuntime( 9285): 	at android.nfc.tech.BasicTagTechnology.connect(BasicTagTechnology.java:78)
E/AndroidRuntime( 9285): 	at android.nfc.tech.Ndef.connect(Ndef.java:71)
E/AndroidRuntime( 9285): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:314)
E/AndroidRuntime( 9285): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime( 9285): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime( 9285): 	at java.lang.Thread.run(Thread.java:818)
E/AndroidRuntime(10762): FATAL EXCEPTION: pool-1-thread-6
E/AndroidRuntime(10762): Process: com.ionicframework.nfclabel717026, PID: 10762
E/AndroidRuntime(10762): java.lang.IllegalStateException: Close other technology first!
E/AndroidRuntime(10762): 	at android.nfc.Tag.setConnectedTechnology(Tag.java:458)
E/AndroidRuntime(10762): 	at android.nfc.tech.BasicTagTechnology.connect(BasicTagTechnology.java:78)
E/AndroidRuntime(10762): 	at android.nfc.tech.Ndef.connect(Ndef.java:71)
E/AndroidRuntime(10762): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:314)
E/AndroidRuntime(10762): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(10762): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(10762): 	at java.lang.Thread.run(Thread.java:818)
E/AndroidRuntime(11514): FATAL EXCEPTION: pool-1-thread-5
E/AndroidRuntime(11514): Process: com.ionicframework.nfclabel717026, PID: 11514
E/AndroidRuntime(11514): java.lang.IllegalStateException: Close other technology first!
E/AndroidRuntime(11514): 	at android.nfc.Tag.setConnectedTechnology(Tag.java:458)
E/AndroidRuntime(11514): 	at android.nfc.tech.BasicTagTechnology.connect(BasicTagTechnology.java:78)
E/AndroidRuntime(11514): 	at android.nfc.tech.Ndef.connect(Ndef.java:71)
E/AndroidRuntime(11514): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:314)
E/AndroidRuntime(11514): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(11514): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(11514): 	at java.lang.Thread.run(Thread.java:818)
E/NfcPlugin(17713): Failed to make tag read only
E/NfcPlugin(17713): java.io.IOException
E/NfcPlugin(17713): 	at android.nfc.tech.Ndef.makeReadOnly(Ndef.java:388)
E/NfcPlugin(17713): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:319)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/NfcPlugin(17713): 	at java.lang.Thread.run(Thread.java:818)
E/NfcPlugin(17713): Failed to make tag read only
E/NfcPlugin(17713): java.io.IOException
E/NfcPlugin(17713): 	at android.nfc.tech.Ndef.makeReadOnly(Ndef.java:388)
E/NfcPlugin(17713): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:319)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/NfcPlugin(17713): 	at java.lang.Thread.run(Thread.java:818)
E/NfcPlugin(17713): Failed to make tag read only
E/NfcPlugin(17713): java.io.IOException
E/NfcPlugin(17713): 	at android.nfc.tech.Ndef.makeReadOnly(Ndef.java:388)
E/NfcPlugin(17713): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:319)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/NfcPlugin(17713): 	at java.lang.Thread.run(Thread.java:818)
E/NfcPlugin(17713): Failed to make tag read only
E/NfcPlugin(17713): java.io.IOException
E/NfcPlugin(17713): 	at android.nfc.tech.Ndef.makeReadOnly(Ndef.java:388)
E/NfcPlugin(17713): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:319)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/NfcPlugin(17713): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/NfcPlugin(17713): 	at java.lang.Thread.run(Thread.java:818)

E/AndroidRuntime(17713): FATAL EXCEPTION: pool-1-thread-5
E/AndroidRuntime(17713): Process: com.ionicframework.nfclabel717026, PID: 17713
E/AndroidRuntime(17713): java.lang.IllegalStateException: Close other technology first!
E/AndroidRuntime(17713): 	at android.nfc.Tag.setConnectedTechnology(Tag.java:458)
E/AndroidRuntime(17713): 	at android.nfc.tech.BasicTagTechnology.connect(BasicTagTechnology.java:78)
E/AndroidRuntime(17713): 	at android.nfc.tech.Ndef.connect(Ndef.java:71)
E/AndroidRuntime(17713): 	at com.chariotsolutions.nfc.plugin.NfcPlugin$2.run(NfcPlugin.java:314)
E/AndroidRuntime(17713): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(17713): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(17713): 	at java.lang.Thread.run(Thread.java:818)`

taglme, 2016-05-11:
May be it depend on NFC chip type?
I get this error while trying to encode ntag216 and ntag213 chip
Link to full log
https://yadi.sk/i/VRqT4xGUreDJx

don, 2016-06-14:
It's possible the tags you have don't support being made read only. (You're supposed to get a better error if that happen.)

Can you debug the Android code and see where it's failing? Start here https://github.com/chariotsolutions/phonegap-nfc/blob/ccd39de398861f83d9f4cc8fa05ffa5ce4ddaaf3/src/android/src/com/chariotsolutions/nfc/plugin/NfcPlugin.java#L314-L322

closed 2018-04-18 by don

Piyali123, 2021-05-01:
I'm getting Java.IO.Exception while connecting ISO 15693 tag in Xamarin Android.PFB code -    

 Tag tag = intent.GetParcelableExtra(NfcAdapter.ExtraTag) as Tag;

                    byte[] id = tag.GetId();

                     nfcvTag = NfcV.Get(tag);

                    int[] intArray = new int[36];

                    // 4d196b8c-aa42-4f90-924f-b953654c31f7 (Length = 36)
                    string guidStr = Guid.NewGuid().ToString();

                    char[] guidChars = guidStr.ToCharArray();

                    for (int i = 0; i < guidChars.Length; i++)
                    {
                        int value = Convert.ToInt32(guidChars[i]);
                        intArray[i] = value;
                    }

                    if (nfcvTag != null)
                    {
 nfcvTag.Connect();