chariotsolutions/phonegap-nfc #235: LF, HF and UHF

jszersze, 2016-05-06:
Hello, 

Is there any way to configure the plugin to be able to read at different frequencies? Currently I'm experiencing a delay in scanning and I suspect that it's do to lower frequency. Also I have to place the tag  I want to scan at a very precise position on the back of the device (currently testing against Samsung S5). Do you have any advice on increasing the scanning speed and distance in relation to the device? 

Thanks!

don, 2016-05-06:
NFC works at 13.56 MHz (like RFID HF). 

Some phones are very finicky about tag placement when scanning. Some tags work better than others on certain phones. Generally I find type 2 tags are faster than type 4 tags.

I don't think there are any options you can configure in the code. There may be Samsung specific options on your phone.

jszersze, 2016-05-17:
Thank you. I tested a bunch more tags and it leads me to another question. I will post it as a separate topic for clarity.

closed 2016-05-17 by don