chariotsolutions/phonegap-nfc #236: Question: Detect if ran from intent vs normal tag read

NoCodeNoMore, 2016-05-09:
Hi, I know how to read tag and launch it from intent.

The question is how do I make difference, if app was launched from Intent or tag was read after app already opened?

don, 2016-05-09:
All tags come from intents, so there's not really a good way to do that.

When the [app is launched by scanning a tag](https://github.com/chariotsolutions/phonegap-nfc#launching-your-android-application-when-scanning-a-tag), you'll receive a ndef-mime event.

If the tags you are scanning a not TNF_MIME_MEDIA, you can use a NDEF listener. This way the MIME listener gets tags launching the app and the NDEF listener gets tags after the app is launched.

NoCodeNoMore, 2016-05-10:
Thank you, I thought about this solution, needed confirmation.

closed 2016-05-17 by don