chariotsolutions/phonegap-nfc #244: nfc.addNdefListener event not fired but success event work

gianvoci, 2016-07-06:
I'm using a nexus 5 + marshmallow S.O. and with latest phonegap 6.2.9 when I try to read a Mifare Forum type 2, with some data, the callback isn't fired.

Here's my piece of code:
```

    onDeviceReady: function() {
        alert('event received');
        nfc.addNdefListener(
            function(nfcEvent) {
                alert('received');
            },
            function() { // success callback
                alert("Waiting for NDEF tag");
            },
            function(error) { // error callback
                alert("Error adding NDEF listener " + JSON.stringify(error));
            }
        );
    }

```

don, 2016-07-06:
Use https://play.google.com/store/apps/details?id=com.nxp.nfc.tagwriter&hl=en to write a new message onto the tag. A text record is fine.

Then use your app to read the tag. The tag must contain an NDEF message for nfc.addNdefListener to fire.

gianvoci, 2016-07-06:
Already tried, with same app and other app, but doens't work.
Other ideas?

don, 2016-07-06:
If you can read and write NDEF with NxP Tag Writer, your phone works OK. If not maybe the tag is bad.

Check your app for JavaScript errors. Watch `adb logcat` output. Connect with chrome://inspect and look for errors.

don, 2016-07-06:
It's possible there is some android-6.0 permission that needs to be added, target android-22 in config.xml to eliminate that issue.

gianvoci, 2016-07-07:
Using inspector give me that error:

[image]

And in log cat:

> 07-07 11:33:29.297 32332 32332 D NfcPlugin: onPause Intent {  }
07-07 11:33:29.297 32332 32332 D NfcPlugin: stopNfc
07-07 11:33:29.298 32332 32332 D NfcPlugin: onNewIntent Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x34000000 cmp=it.ekra.puntoit/.MainActivity (has extras) }
07-07 11:33:29.298 32332 32332 D NfcPlugin: onResume Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x34000000 cmp=it.ekra.puntoit/.MainActivity (has extras) }
07-07 11:33:29.298 32332  2212 D NfcPlugin: parseMessage Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x34000000 cmp=it.ekra.puntoit/.MainActivity (has extras) }
07-07 11:33:29.298 32332  2212 D NfcPlugin: action android.nfc.action.TECH_DISCOVERED
07-07 11:33:29.298 32332  2212 D NfcPlugin: android.nfc.tech.NfcA
07-07 11:33:29.298 32332  2212 D NfcPlugin: android.nfc.tech.MifareUltralight
07-07 11:33:29.298 32332  2212 D NfcPlugin: android.nfc.tech.Ndef
07-07 11:33:29.300  2874 29395 W AudioTrack: AUDIO_OUTPUT_FLAG_FAST denied by client; transfer 4, track 16000 Hz, output 48000 Hz
07-07 11:33:29.301 32332  2212 V NfcPlugin: var e = document.createEvent('Events');
07-07 11:33:29.301 32332  2212 V NfcPlugin: e.initEvent('ndef');
07-07 11:33:29.301 32332  2212 V NfcPlugin: e.tag = {"id":[4,-58,106,-126,-70,41,-128],"techTypes":["android.nfc.tech.NfcA","android.nfc.tech.MifareUltralight","android.nfc.tech.Ndef"],"type":"NFC Forum Type 2","maxSize":142,"isWritable":true,"ndefMessage":[{"tnf":1,"type":[84],"id":[],"payload":[2,105,116,80,114,111,118,97,32,49,50,51]}],"canMakeReadOnly":true};
07-07 11:33:29.301 32332  2212 V NfcPlugin: document.dispatchEvent(e);
07-07 11:33:29.304 32332 32332 I chromium: [INFO:CONSOLE(1070)] "Uncaught EvalError: Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "default-src * 'unsafe-inline'".
07-07 11:33:29.304 32332 32332 I chromium: ", source: file:///android_asset/www/cordova.js (1070)
07-07 11:33:29.307  2874  2362 D NativeNfcTag: Starting background presence check
07-07 11:33:29.593 18466 18507 I EffectDs: Effect_process(b6002500) called on session 0 (logs suppressed for next second) (inBuffer==outBuffer)?TRUE! outputConfig.accessMode=WRITE
07-07 11:33:30.461   866 23322 D NetlinkSocketObserver: NeighborEvent{elapsedMs=447658655, 192.168.0.1, [4C60DE816DF0], RTM_NEWNEIGH, NUD_STALE}
07-07 11:33:30.622 18466 18507 I EffectDs: Effect_process(b6002500) called on session 0 (logs suppressed for next second) (inBuffer==outBuffer)?TRUE! outputConfig.accessMode=WRITE
07-07 11:33:30.646  2874  2362 D NativeNfcTag: Tag lost, restarting polling loop
07-07 11:33:30.646  2874  2362 D NfcService: Discovery configuration equal, not updating.
07-07 11:33:30.646  2874  2362 D NativeNfcTag: Stopping background presence check

gianvoci, 2016-07-07:
I'v discovered the issue, is in my html:

<meta http-equiv="Content-Security-Policy" content="default-src * 'unsafe-inline'; style-src 'self' 'unsafe-inline'; media-src *" />

removing that I'v fixed the issue, but, you know why? My html is from hello world demo.

don, 2016-07-07:
Glad it's working. The content security policy blocked your inline JavaScript eval. Tough to tell exactly what the problem is without seeing your code or the sample you used.

gianvoci, 2016-07-07:
I'v used the helloworld created by CLI "phonegap create ....." etc.:

`<!DOCTYPE html>
<html>

<head>
    <meta charset="utf-8" />
    <meta name="format-detection" content="telephone=no" />
    <meta name="msapplication-tap-highlight" content="no" />
    <meta name="viewport" content="user-scalable=no, initial-scale=1, maximum-scale=1, minimum-scale=1, width=device-width" />
      WITHOUT THIS NOW WORK
    <link rel="stylesheet" type="text/css" href="css/index.css" />
    <title>Test NFC</title>
</head>

<body>
    <center>NFC Demo</center>
    <script type="text/javascript" src="cordova.js"></script>
    <script type="text/javascript" src="js/index.js"></script>
    <script type="text/javascript">
        app.initialize();
    </script>
</body>

</html>`

and:

`var app = {
    // Application Constructor
    initialize: function() {
        this.bindEvents();
    },
    // Bind Event Listeners
    //
    // Bind any events that are required on startup. Common events are:
    // 'load', 'deviceready', 'offline', and 'online'.
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
    },
    // deviceready Event Handler
    //
    // The scope of 'this' is the event. In order to call the 'receivedEvent'
    // function, we must explicitly call 'app.receivedEvent(...);'
    onDeviceReady: function() {
        nfc.addNdefListener(
            function(nfcEvent) {
                var tag = nfcEvent.tag,
                    ndefMessage = tag.ndefMessage;

                // dump the raw json of the message
                // note: real code will need to decode
                // the payload from each record
                alert(JSON.stringify(ndefMessage));

                // assuming the first record in the message has
                // a payload that can be converted to a string.
                alert(nfc.bytesToString(ndefMessage[0].payload).substring(3));
            },
            function() { // success callback
                alert("Waiting for NDEF tag");
            },
            function(error) { // error callback
                alert("Error adding NDEF listener " + JSON.stringify(error));
            }
        );
    }
};
`

nothing all.

closed 2016-07-13 by don