chariotsolutions/phonegap-nfc #246: How to format a tag when tag is formatable

rajvaida, 2016-07-08:
The card I am reading is not pre-formated. But it's NdefFormatable is in techTypes. How to format and get tag data/Payload?

don, 2016-07-08:
Use [nfc.addNdefFormatableListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddndefformatablelistener) (or perhaps [nfc.addTagDiscoveredListener](https://github.com/chariotsolutions/phonegap-nfc#nfcaddtagdiscoveredlistener).)

You can read the tag meta data when it's discovered. There's no payload when you use nfc.addNdefFormatableListener or nfc.addTagDiscoveredListener.

To format the tag, use [nfc.write](https://github.com/chariotsolutions/phonegap-nfc#nfcwrite) and write an empty message onto the tag.

    // This code belongs in the success callback for nfc.addNdefFormatableListener
    // write an empty NDEF record to format the tag
    var message = [
        ndef.emptyRecord();
    ];

    nfc.write(message, [onSuccess], [onFailure]);

rajvaida, 2016-07-08:
Thank you. I am trying to implement something similar to https://github.com/codebutler/farebot

closed 2016-07-08 by rajvaida

don, 2016-07-08:
It looks like [farebot](https://github.com/codebutler/farebot) reads the raw bytes from the NFC tags. This plugin can only read NDEF messages. If you want to support the same cards as farebot, you're probably better off creating a new plugin that wraps the farebot code (assuming you're OK with [GPL](https://github.com/codebutler/farebot/blob/master/COPYING).). The plugin could send the [card XML](https://github.com/codebutler/farebot/blob/5aa939f750e27b21d8ea53d72ee4a0e298629845/src/main/java/com/codebutler/farebot/activity/ReadingTagActivity.java#L86-L88) to javascript for processing.

Lughino, 2016-11-03:
I have a problem to format a `ndef-formatable` tag with the procedure you described.
Every time I try to write the error callback is called with the argument error to null.
I tried to write a card with an external application (NXP TagWrite) and is written correctly. After having written the card is properly recognized by your plugin through `addNdefListener` and I can read and write.
This is the code that I use to test the writing:
```javascript
window.nfc.addNdefFormatableListener(
        function(nfcData) {
          console.log('data in formatable', nfcData);
          window.nfc.write([
              window.ndef.emptyRecord()
            ],
            function(d) {
              console.log('success', d);
            },
            function(e) {
              console.log('error', e, arguments); // this is always null
            }
          );
        },
        function() {
          console.log('listen nfc formatable');
        },
        function(err) {
          console.log(err);
        }
      );
```

This is the content of the card:
```javascript
{
bubbles: false,
cancelBubble: false,
cancelable: false,
clipboardData: undefined,
currentTarget: null,
defaultPrevented: false,
eventPhase: 0,
returnValue: true,
srcElement: document,
tag: {
  id: [4, 94, -71, -54, -87, 74, -127],
  techTypes: ["android.nfc.tech.MifareUltralight", "android.nfc.tech.NfcA", ]
},
target: document,
timeStamp: 1478188522177,
type: "ndef-formatable"
}
```

How can I fix the problem?
Thank you

RicardoLopes91, 2018-12-05:
Hi @don 

I have the same problem as @Lughino.
Is there any way to solve this?

Thank you.