chariotsolutions/phonegap-nfc #247: I am not able to retrieve NFC tags

jwillmer, 2016-07-26:
This function:

```
nfc.addTagDiscoveredListener(function (nfcEvent) {
    console.log("discovered");
    console.log(JSON.stringify(nfcEvent));
}, function () {
    console.log("Discovering NFC Messages.");
},
function (error) {
    console.log("Failed to discover NDEF.");
});
```

outputs this:

```
{"isTrusted":false,"tag":{"id":[91,7,-33,-4],"techTypes":["android.nfc.tech.NfcA","android.nfc.tech.MifareClassic","android.nfc.tech.NdefFormatable"]}}
```

but the listener never fires a event:

```
nfc.addNdefListener(function (nfcEvent) {
    console.log("listener");
    console.log(JSON.stringify(nfcEvent));
}, function () {
    console.log("Listening for NFC Messages.");
},
function (error) {
    console.log("Failed to add NDEF listener.");
});
```

I tried to use `addMimeTypeListener` with one of this MIME types (`android.nfc.tech.NfcA`,`android.nfc.tech.MifareClassic`,`android.nfc.tech.NdefFormatable`) but I also don't get a result.

My test NFC card contains a number (plain text). Something like: `123456789`

don, 2016-07-26:
What phone are you using?

Based on the output above, you're scanning a Mifare Classic tag.  Newer phones with Broadcom NFC chips don't support Mifare Classic tags for NDEF.

jwillmer, 2016-07-26:
@don I use a `Xiaomi Mi 5` but I cannot find any specs that tell me which NFC chip it has :(

jwillmer, 2016-07-26:
Android Version `6.0`

don, 2016-07-26:
Maybe it's a permission problem. I haven't updated this plugin for Android 6 permission yet. Can you edit config.xml and target Android-22?

    <platform name="android">
        <allow-intent href="market:*" />
        <preference name="android-targetSdkVersion" value="22" />
    </platform>

jwillmer, 2016-07-26:
If I change it in the project `config.xml`
```
 <platform name="android">
    <allow-intent href="market:*" />
    <preference name="android-targetSdkVersion" value="22" />
  </platform>
  <platform name="ios">
     ...
```

 I get the following error: 
`Failure [INSTALL_FAILED_PERMISSION_MODEL_DOWNGRADE]`

don, 2016-07-26:
That error means you need to delete the app from your phone before installing the new one.

jwillmer, 2016-07-26:
I updated the comment - sorry, was to slow ;)

jwillmer, 2016-07-26:
FYI this [reader on GitHub](https://github.com/nadam/nfc-reader) is available in the Play Store and can display the value of the NFC card.

jwillmer, 2016-07-26:
Maybe your are missing references to `Mifare`?

@nadams source file:
https://github.com/nadam/nfc-reader/blob/master/src/se/anyro/nfc_reader/TagViewer.java#L37-L38

don, 2016-07-26:
That Java code has worked with Mifare for a long time. It's unlikely that is the problem. 

What happened after you installed the android-22 version of the app? Attach more logs from adb if it's still failing.

don, 2016-07-26:
@jwillmer try this sample project https://github.com/don/phonegap-nfc-issue-247

For phones that support Mifare Classic you'll get NDEF data. For other phones like Nexus 4 or Samsung S4 that don't support Mifare Classic, you'll get some tag metadata. Both screenshots below are from the same app scanning the same MifareClassic NFC tag.

Nexus 5X

[image]

Samsung S4

[image]

jwillmer, 2016-07-27:
I tried your solution and the output is this;
[image]

What I don't undeerstand is why another NFC app can display the result/number:
[image]

jwillmer, 2016-07-27:
> That Java code has worked with Mifare for a long time. It's unlikely that is the problem.

I don't say your code is bugy but your code might not cover all cases. I am not a Java programer but for me it looks like that [the code](https://github.com/nadam/nfc-reader/blob/master/src/se/anyro/nfc_reader/TagViewer.java#L143) of @nadam gets more information out of the tags. Wdyt?

don, 2016-07-27:
According to your screenshot, your tag is NdefFormatable, meaning there's no NDEF message on the tag.

The nfc.addNdefListener will only be called when there's an NDEF message on the tag. You need to use nfc.addNdefFormatableListener or nfc.addTagDiscoveredListener to get an event for non-NDEF tags.

Use [NXP TagInfo](https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en) to view the data on the tag.

I updated the demo app to write a message to the tag when it sees a formatable tag. https://github.com/don/phonegap-nfc-issue-247/commit/441314813ed21a91105de338b0f21b5519953884

Alternatively you can write at text record to the tag using [NXP TagWriter](https://play.google.com/store/apps/details?id=com.nxp.nfc.tagwriter&hl=en).

jwillmer, 2016-07-28:
O.K. but there is still a number on the card I need to read and I am not able to get it with that plugin. Is that intended? For me it is a problem because my app needs to read this cards - not format them.

jwillmer, 2016-07-28:
I was blind. I found the Id in the result. I needed to format the id to hex then I have it (reversed).

[image]

closed 2016-07-28 by jwillmer

jwillmer, 2016-07-28:
@don Thank you for your help, you kept me going :+1:

don, 2016-07-28:
You're welcome. Since you only need the ID, you can use the tagDiscoveredListener in your app. It should find any tag.

riebschlager, 2016-08-25:
I'm having a similar issue. I am using `tagDiscoveredListener` and it never seems to be fired. The tablet makes a sound like it is reading successfully, but the `tagDiscoveredListener` callback is never called. The tags I'm trying to scan are `MifareUltralight`.

Any idea what might be happening?

jwillmer, 2016-08-25:
@riebschlager can you open another issue? This one is closed.