chariotsolutions/phonegap-nfc #248: Plugin exception on Windows Phone

jwillmer, 2016-07-27:
I have a Windows Phone 950 and compiled my app as a Windows Phone (universal) app. I get the following error then I try to scan a NFC card.

[image]

I used the same card as in issue #247 and I also deployed/tested the project to an Android without having this problem.

jwillmer, 2016-07-27:
Foud the issue. In Windows Phone the `tag.id` is empty and my convertion

```
var hex = nfc.bytesToHexString(nfcEvent.tag.id)
```

fails.

closed 2016-07-27 by jwillmer

don, 2016-07-27:
Newer Windows devices with certain NFC chipsets can get the tag id. Other devices only get the NDEF data from the tag and id will be null.