chariotsolutions/phonegap-nfc #249: CSP restricts bridge between phonegap and events

alexbarnsley, 2016-07-28:
I'm not sure if this is a directly related issue to #196 . Spent a good few hours trying to figure out why my NFC wasn't being detected (even though I'd used the exact JS elsewhere successfully). I realised that the Content-Security-Policy restricts any events from being fired (from the `this.webView.sendJavascript(command);` command).

Would it be possible to add something to the readme to help others who may experience similar issues in the future? My workaround was to allow all for now.

Does anyone know what the correct script security value should be in this instance?

Thanks.

don, 2016-07-28:
I'll update the docs, but I need more details. 

What did you do to get it to work? Do you have an example of a change to break CSP and what to add to make it work?

alexbarnsley, 2016-08-01:
Sorry for slow reply. The CSP that was causing the issue was 
```
<meta http-equiv="Content-Security-Policy" content="default-src * 'unsafe-inline'; style-src 'self' 'unsafe-inline'; media-src *; img-src * data:" />
```
Because it's internal, to fix I effectively removed all CSP as wasn't too bothered about security.

After some testing just now, I've found that I needed 'unsafe-eval', so my final CSP is now:

```
<meta http-equiv="Content-Security-Policy" content="default-src * 'unsafe-inline' 'unsafe-eval'; style-src 'self' 'unsafe-inline'; media-src *; img-src * data:" />
```

Since reporting this issue, I've found the Chrome Remote Inspection tool which does highlight this as the issue:

>Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "default-src * 'unsafe-inline'". Note that 'script-src' was not explicitly set, so 'default-src' is used as a fallback.
>Uncaught EvalError: Refused to evaluate a string as JavaScript because 'unsafe-eval' is not an allowed source of script in the following Content Security Policy directive: "default-src * 'unsafe-inline'".

I hope this helps.

Thanks,

Alex.

don, 2018-06-01:
Fixed by #323

closed 2018-06-01 by don