chariotsolutions/phonegap-nfc #251: tagDiscoveredListener callback never fires

riebschlager, 2016-08-25:
I'm having a similar issue to #247  I am using tagDiscoveredListener and it never seems to be fired. The tablet makes a sound like it is reading successfully, but the tagDiscoveredListener callback is never called. The tags I'm trying to scan are MifareUltralight. My android version is 4.4.4.

Any idea what might be happening?

don, 2016-08-25:
Use jhint to make sure there are no JavaScript errors in your code.

Try running this sample project https://github.com/don/phonegap-nfc-issue-247 and see if you can scan your tag.

riebschlager, 2016-08-26:
After running that project, it successfully alerts after scanning a tag that I've written an NDEF message to, but any other tag won't work. Again, I hear the sound that it's been read, but there's no alert in the app.

riebschlager, 2016-08-26:
Woah, hold the phone.

If I get rid of every other listener in that app other than `addTagDiscoveredListener`, I can successfully get a read from an unformatted tag. Which is exactly what I thought I had done in my own app. But this is a great lead, thanks!

closed 2016-09-17 by don