chariotsolutions/phonegap-nfc #255: Cannot launch app on Android when scanning a tag

bastos71, 2016-09-28:
Hi everyone !

I'm trying to automaticaly launch my app when scanning a NFC tag.
I don't understand why but it fails :/

I saw on Readme.md I have to add thos lines in my /platforms/android/AndroidManifest.xml :

`<intent-filter>`
`    <action android:name="android.nfc.action.NDEF_DISCOVERED" />`
`    <data android:mimeType="text/pg" />`
`    <category android:name="android.intent.category.DEFAULT" />`
`</intent-filter>`

Could someone give me a config.xml file example to understand my mistake ? 

Thanks !

don, 2017-01-05:
You need to add the intent filter into AndroidManifest.xml not config.xml

Change android:mimeType from text/pg to match the mime type of your tag. You can also add different intent filters based on the tag content. See https://developer.android.com/guide/topics/connectivity/nfc/nfc.html#ndef-disc

Note that you need ndef.addMimeTypeListener in your app to handle the tag that is scanned to launch the app.

closed 2017-01-05 by don

don, 2017-01-05:
@bastos71 this after_prepare hook works well for adding an intent to the AndroidManifest.xml http://stackoverflow.com/a/35597489/1416463

bastos71, 2017-01-09:
Thank you for your help @don , I will take a look as soon as possible.