chariotsolutions/phonegap-nfc #257: remove event listener not working

jszersze, 2016-11-01:
It looks like the 'removeTagDiscoveredListener' is not removing the function that is being triggered by the event listener. When I try to overwrite the function (as a work around) and set it to an empty function, `myfunc = function(){}` it still gets triggered also. It's almost like it makes another internal copy of this function that becomes unavailable upon event listeners creation. 

Is there any solutions to getting rid of either the event of the function triggered?

don, 2017-01-05:
Can you create a simple example to demonstrate this problem? That would help me debug the issue.

jeremycoppey, 2017-10-17:
I'm experiencing the same behavior here, the remove doesn't work. did you already worked on this issue? You need some extra information?

PR #322 (2018-05-31): remove listener issue 257
Fixes #257
commit 77bf9189 (2018-05-31): Merge pull request #322 from chariotsolutions/issue-257

remove listener fixes #257
closed 2018-05-31 by don