chariotsolutions/phonegap-nfc #259: addTagDiscoveredListener event fired not fired from time to time

moto-a, 2016-12-08:
I got cordova project fro android 4.1.1.
AddTagDiscoveredListener event not fired from time to time.
I think that if nfc service stop, callback function is lost.
"local reference table overflow " is causing a memory leak?
I'd like to tell me how to solve.

The logs are as follows.
12-08 13:06:16.278 25737-25753/? E/dalvikvm: JNI ERROR (app bug): local reference table overflow (max=512)
12-08 13:06:16.793 128-128/? I/ServiceManager: service 'nfc' died
12-08 13:06:16.793 396-23376/? I/ActivityManager: Process com.android.nfc (pid 25737) has died.
12-08 13:06:16.809 396-23376/? I/ActivityManager: Start proc com.android.nfc for restart com.android.nfc: pid=27412 uid=1027 gids={3002, 3001, 1015, 1028}
12-08 13:06:16.848 27412-27412/? I/NfcService: Starting NFC service
12-08 13:06:16.856 27412-27412/? D/NFCJNI: NFC Service : loading JNI
12-08 13:06:16.895 27412-27412/? I/NfceeAccess: read 0 signature(s) for NFCEE access
12-08 13:06:16.957 27412-27426/? D/NfcService: checking on firmware download
12-08 13:06:16.957 27412-27426/? D/NfcService: NFC is on. Doing normal stuff
12-08 13:06:16.957 27412-27426/? I/NfcService: Enabling NFC
12-08 13:06:16.957 27412-27426/? D/NFCJNI: Start Initialization

don, 2017-01-05:
It looks like the Android NFC stack is crashing.

     JNI ERROR (app bug): local reference table overflow (max=512)
     service 'nfc' died

Does this happen on more than one phone?
Can you reliably reproduce the bug with your NFC tags or app?
Does scanning your tags with different apps cause the same problem?

moto-a, 2017-01-07:
Hi don, Thank you for your reply.

>Does this happen on more than one phone?
Yes. This issue is happen another phone too.

>Can you reliably reproduce the bug with your NFC tags or app?
This issue does not always occur, but it happens quite often.
Frequent occur when cards are read close to the level at which sound does not come out or released.

>Does scanning your tags with different apps cause the same problem?
Sorry,I can not try it.

don, 2017-01-07:
I expect this is an Android bug, but it could be something in the plugin, the tags contents, or the app.

Can you share a simple app that crashes when scanning tags? That would allow me to try and duplicate the problem to track down the issue.

moto-a, 2017-01-07:
[www.zip](https://github.com/chariotsolutions/phonegap-nfc/files/691377/www.zip)
I send sample app.
I think that this issue is happen Android 4.1.1 or 4.2 bug.

moto-a, 2017-03-06:
I think that this issue is happen Android 4.1.1.
Please refer to the following URL.
https://code.google.com/p/android/issues/detail?id=54621

I am trying to do the following to solve this problem.
First, Broadcast Intent notified when NfcService restarts
Listening on the Broadcast Receiver and processing.
Second, NfcAdapter.isEnabled ,NfcAdapter.disableForegroundDispatch and NfcAdapter.enableForegroundDispatch.

I send pilot code.
[NfcPlugin.zip](https://github.com/chariotsolutions/phonegap-nfc/files/819900/NfcPlugin.zip)

don, 2018-04-18:
Closing old issue. Hoping new Cordova and Android fix this.