chariotsolutions/phonegap-nfc #264: unable to detect NFC tag in Ionic2 with Meteor1.4

tomitbe, 2017-01-05:
Tried to integrate the phonegap-nfc in Meteor with Ionic2 (using the angular-meteor boilerplate).
The NFC module is available (I can for instance open the showSettings() method with success from inside my app) but the listeners don't get triggered.
What am I doing wrong? I tried to find out how to detect any tag in Android:

export class AppComponent {
  rootPage: any;  
  
  constructor(platform: Platform,translate: TranslateService) {   
    
    this.rootPage = Meteor.user() ? navComponent : LoginComponent;        
    
    platform.ready().then(() => {
    
      NFC.addTagDiscoveredListener('text/plain',(nfcEvent) => {  
                console.log("success" + JSON.stringify(nfcEvent.tag, null, 4));               
            }, (fail) => {
                console.log("failed for NDEF Tags." + fail);
            });

      StatusBar.styleDefault();
    });
  }
}

not sure about the first mime-type argument, but when I scan my tag via the built-in android nfc-tool, I have a value "test" in it in text/plain format. also tried with "addNdefListener" but without success.

Would really appreciate your help on this,

thanks!

don, 2017-01-06:
I don't know meteor. Can you create a simple project and commit it to github? Add the link here and I'll take a look.

tomitbe, 2017-01-06:
Hi Don, Thanks for your prompt reply. I have created a repo on
https://github.com/tomitbe/cignos

you can find the nfc.component.ts file here (where I add the eventListeners):
https://github.com/tomitbe/cignos/blob/master/client/imports/pages/nfc/nfc.component.ts

to test the app yourself, install Meteor (latest 1.4), clone my repo, inside the folder, run npm --install and then "meteor run android-device" to test it.

tomitbe, 2017-01-06:
Hi Don,

I have found a working solutions! 
https://forum.ionicframework.com/t/nfc-ionic-v2-not-work/67490

the problem was that I didn't subscribe the observable function addNdefListener() 

thanks anyway for your help!

tomitbe, 2017-01-07:
Hi Don, just another question... how can I get the NFC structure to something readable? 
NFC.bytestostring() is not included in the Ionic package.
Though I want to be able to read the ID of the NFC tag and also its content:

{"isTrusted":false,"tag":{"isWritable":true,"id":[4,127,-71,66,112,64,-128],"techTypes":["android.nfc.tech.MifareUltralight","android.nfc.tech.NfcA","android.nfc.tech.Ndef"],"type":"NFC Forum Type 2","canMakeReadOnly":true,"maxSize":137,"ndefMessage":[{"id":[],"type":[97,110,100,114,111,105,100,46,99,111,109,58,112,107,103],"payload":[111,114,103,46,110,100,101,102,116,111,111,108,115,46,98,111,105,108,101,114,112,108,97,116,101],"tnf":4},{"id":[],"type":[84],"payload":[2,101,110,67,117,115,116,111,109,119,101,98],"tnf":1}]}

tomitbe, 2017-01-11:
Hi Don, still struggling with reading the id (eg [4,127,-71,66,112,64,-128]) => how to convert this to a normal unique number? 2nd, in Ionic2 I am missing the removeNdefListener() method on the NFC object. Any advise on how to kill an earlier created listener?

don, 2017-02-22:
@tomitbe use [nfc.bytesToHexString](https://github.com/chariotsolutions/phonegap-nfc/blob/master/www/phonegap-nfc.js#L598) to turn this back into a printable id.

IMO removing listeners is bad. I always leave the listeners in place. For cases where I don't need the data, I have the listener ignore incoming data rather than ignore it.

If you really want remove listeners, the functions are available in the plugin https://github.com/chariotsolutions/phonegap-nfc#nfcremovendeflistener. You'd just need to add code in the Ionic wrapper https://ionicframework.com/docs/v2/native/nfc/ or call the plugin code directly.  You should be able to declare `declare var nfc: any;` and call this plugin directly instead of the Ionic wrapper `NFC`.

closed 2018-04-18 by don