chariotsolutions/phonegap-nfc #266: Doesn't read Oyster card

rolandkofler, 2017-01-19:
The London subway card "Oyster card" is not read, what are the possible causes?

nestorsilk, 2017-02-11:
Oyster card is a mifare desire ev1 card. Works differently. But in case you intend to work with desfire cards , especially in block chain context ( I assume) you can contact me.

don, 2017-02-12:
This plugin reads and writes NDEF messages on tags. Mifare DESfire tags with NDEF messages should work fine, however like @nestorsilk mentioned that Oyster card is different. 

You can probably use https://play.google.com/store/apps/details?id=com.nxp.taginfolite&hl=en to view the card contents. The Android SDK should have the classes you need to read and write sectors.

closed 2017-02-12 by don