chariotsolutions/phonegap-nfc #269: Petition for the inclusion of a method such as registerNFCStateChangeHandler()

fjms, 2017-02-21:
To verify if nfc is enabled, I resort to catch NFC.enabled (). This is useful if the NFC is disabled when starting the app. But if the user (who has no idea) disable the NFC at some point in the app I can not capture that event. The only thing I can think of is having a setInterval() in the background checking NFC.enabled.

I would appreciate the development of a listener that will listen when the state of the NFC changes, just like the Cordova Diagnostic plugin.

Thank.

closed 2017-02-22 by fjms

don, 2017-02-22:
@fjms Did you come up with an alternate solution?

fjms, 2017-02-22:
The functionality existed already https://github.com/dpa99c/cordova-diagnostic-plugin#registernfcstatechangehandler :P 
I did not see it when I looked at the diagnostic plugin.

don, 2017-02-22:
Excellent, thanks for the link