chariotsolutions/phonegap-nfc #271: Capable of reading credit card info?

AshleyDeng, 2017-03-17:
Hi, 

Thanks for making this plugin! It looks like this repo is very well managed. Keep up the good work!

I am wondering if we could read credit card info (card number, expiry date, and even cvv) with this plugin? 

Currently, I could use `addTagDiscoveredListener` to read the tag ID (4 bytes) and tech types (`android.nfc.tech.IsoDep, android.nfc.tech.NfcA`) from a cc. Since the cc doesn't have NDEF message stored so I don't suppose to (and didn't) get any call back from `addNdefListener`.

Any help is very much appreciated! 
Ash

don, 2017-03-17:
This plugin just reads NDEF. You *might* be able to use Host Card Emulation (HCE) to read your credit card. Check out https://github.com/don/cordova-plugin-hce

closed 2017-03-17 by don

kapilkarda, 2018-03-26:
is their anything available for android and ios ?