chariotsolutions/phonegap-nfc #272: UWP Tag Id

andreujuanc, 2017-03-21:
Was checking https://github.com/chariotsolutions/phonegap-nfc/issues/241 and I found that for Windows there is a way, I'm about to implement myself, but I would prefer to post it here and start working together, since it's not easy task.

The idea is to use Windows.Devices.SmartCards instead of Windows.Networking.Proximity.ProximityDevice and then PCSC SDK, which already implements all formatting and communication for different cards.

I got it working , more or less. I did port some code from C# to WinJS, but it is LOTS of work. Maybe I would prefer to reference a DLL instead.

Even just having a separate plugin just to read the ID on Windows would be nice, since it's a handy feature.

PCSC SDK: https://github.com/springcard/springcard.pcsc.sdk
Sample code from Msft: https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/Nfc

Ideas?

Cheers

don, 2017-05-04:
Full smartcard support probably deserves it's own plugin. It would be nice to get the UID on Windows if Windows.Devices.SmartCards. Does Windows.Devices.SmartCards support NDEF too? If you have some code I'm willing to look at it and consider pull requests.

andreujuanc, 2017-06-13:
I found that not all Win10 devices are required to support smartcard. For example my Lumia 950 does but I work with some Honeywell's CT50 an, even if they have Win10 device, don't support SmartCard thus no RF communication  > no ids. :( 

If Microsoft forced NFC drivers to be 100% Win10 API implemented would be great. But for now it's hard to know.

I will close it now, if in the future I find anything, i will reopen or PR any good solution for this.

Cheers

closed 2017-06-13 by andreujuanc