chariotsolutions/phonegap-nfc #274: Android | Unexpected behaviour |  ERROR Class not found

andreujuanc, 2017-06-03:
Cordova 7.0.1
Cordova Android Platform: 6.2.3
Android (SO): 6.0.1

1. Scan Tag > Device sound > `ERROR Class not found` > No events are fired
2. Scan Tag > Device sound > Event is Fired > `ERROR Class not found`  is logged less than a second after

I do not have another android phone to test this for now.

I could understand that ERROR Class not found is logged when plugin is not correctly installed/built, but what's very estrange is that it's just start working on the second try. (same run)

How can I debug this? 

https://pastebin.com/6Ue24s9h

andreujuanc, 2017-06-03:
Just compared logs. First first scan. Second is right after that one.

Looks like the intents are different for some reason.

```
06-02 20:13:47.748  3756  3756 D NfcJni  : nativeNfcTag_doReconnect: exit 0x0
06-02 20:13:47.751  1948  3893 I ActivityManager: START u0 {act=android.nfc.action.NDEF_DISCOVERED typ=application/nubila flg=0x24000000 cmp=com.this.is.my.app.name/.MainActivity (has extras)} from uid 10105 on display 0
06-02 20:13:47.751  1948  3893 W ActivityManager: startActivity called from non-Activity context; forcing Intent.FLAG_ACTIVITY_NEW_TASK for: Intent { act=android.nfc.action.NDEF_DISCOVERED typ=application/nubila flg=0x24000000 cmp=com.this.is.my.app.name/.MainActivity (has extras) }
06-02 20:13:47.786 18824 18867 V NfcPlugin: var e = document.createEvent('Events');
06-02 20:13:47.786 18824 18867 V NfcPlugin: e.initEvent('ndef-mime');
06-02 20:13:47.786 18824 18867 V NfcPlugin: e.tag = {"id":[4,-25,-54,98,34,75,-128],"techTypes":["android.nfc.tech.NfcA","android.nfc.tech.MifareUltralight","android.nfc.tech.Ndef"],"type":"NFC Forum Type 2","maxSize":137,"isWritable":true,"ndefMessage":[{"tnf":2,"type":[97,112,112,108,105,99,97,116,105,111,110,47,110,117,98,105,108,97],"id":[],"payload":[110,99,89,86,118,79,70,86,102,80,99,52,55,113,56,119,99,111,106,122,74,48,43,116,43,49,83,119,88,88,105,86,100,115,122,114,107,69,65,74,43,66,111,97,97,119,61,61]}],"canMakeReadOnly":true};
06-02 20:13:47.786 18824 18867 V NfcPlugin: document.dispatchEvent(e);
06-02 20:13:47.788 18824 18920 D PluginManager: exec() call to unknown plugin: barcode
06-02 20:13:47.791 18824 18920 D PluginManager: exec() call to unknown plugin: barcode
06-02 20:13:47.797 18824 18824 D SystemWebChromeClient: file:///android_asset/www/app/services/readerService.js: Line 127 : ERROR
06-02 20:13:47.798 18824 18824 I chromium: [INFO:CONSOLE(127)] "ERROR", source: file:///android_asset/www/app/services/readerService.js (127)
06-02 20:13:47.912  3756 19196 D NfcJni  : nativeNfcTag_doPresenceCheck

```

```

06-02 20:24:37.459  3756  3756 D NfcJni  : nativeNfcTag_doReconnect: exit 0x0
06-02 20:24:37.462  1948  3403 I ActivityManager: START u0 {act=android.nfc.action.TECH_DISCOVERED flg=0x24000000 cmp=com.this.is.my.app.name/.MainActivity (has extras)} from uid 10105 on display 0
06-02 20:24:37.462  1948  3403 W ActivityManager: startActivity called from non-Activity context; forcing Intent.FLAG_ACTIVITY_NEW_TASK for: Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x24000000 cmp=com.this.is.my.app.name/.MainActivity (has extras) }
06-02 20:24:37.464 18824 18824 D CordovaActivity: Paused the activity.
06-02 20:24:37.465 18824 18824 D NfcPlugin: onPause Intent {  }
06-02 20:24:37.465 18824 18824 D NfcPlugin: stopNfc
06-02 20:24:37.467 18824 18824 D NfcPlugin: onNewIntent Intent { act=android.nfc.action.TECH_DISCOVERED flg=0x34000000 cmp=com.this.is.my.app.name/.MainActivity (has extras) }
06-02 20:24:37.469 18824 18824 D CordovaActivity: Resumed the activity.
06-02 20:24:37.617  3756 20307 D NfcJni  : nativeNfcTag_doPresenceCheck
```

don, 2017-07-06:
@andreujuanc is the first scan launching the app and the 2nd scan after the app is open?

If you launch the app with an intent in the manifest, the first scan is always ndef-mime so you need to add a 2nd listener to handle the case. It's find to register the same function to handle both  the NDEF and mime scans.

closed 2018-04-18 by don

andreujuanc, 2018-04-18:
@don  Sorry i left this issue open. It's been a long ride with WM and we already ported all of our custom code and moved customer devices to android. :(