chariotsolutions/phonegap-nfc #281: iOS build failes in PGB

mgoetz85, 2017-09-22:
Dear all,

I am trying to developed a very simple prototype for iOS, however PGB always throws an error.
If I remove the plugin / config line

`<plugin spec="https://github.com/chariotsolutions/phonegap-nfc.git" source="git" />`

the build passes successfully.

App-ID is 2814573.

`The following build commands failed:
	CompileC /Users/phonegap/Library/Developer/Xcode/DerivedData/NFC_Test-eeofxxwbgnwiqtdcybqstaxloonf/Build/Intermediates/ArchiveIntermediates/NFC\ Test/IntermediateBuildFilesPath/NFC\ Test.build/Release-iphoneos/NFC\ Test.build/Objects-normal/armv7/NfcPlugin.o NFC\ Test/Plugins/phonegap-nfc/NfcPlugin.m normal armv7 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
Error code 65 for command: xcodebuild with args: -xcconfig,/private/tmp/gimlet/2814573/project/cordova/build-release.xcconfig,-workspace,NFC Test.xcworkspace,-scheme,NFC Test,-configuration,Release,-destination,generic/platform=iOS,-archivePath,NFC Test.xcarchive,archive,CONFIGURATION_BUILD_DIR=/private/tmp/gimlet/2814573/project/build/device,SHARED_PRECOMPS_DIR=/private/tmp/gimlet/2814573/project/build/sharedpch`

Any idea?

don, 2017-09-22:
I’d guess that PhoneGap Build can’t handle NFC on iOS yet. iOS NFC support needs Xcode9 which was just released this week. Typically it takes PGB a bit to implement new feature.

zxfrank, 2017-10-20:
I also get an error

phonegap-nfc/NfcPlugin.h:11:9: fatal error: 'CoreNFC/CoreNFC.h' file not found
#import <CoreNFC/CoreNFC.h>

@mgoetz85 do you have this error too or different?

felipel90, 2017-10-20:
+1 @zxfrank

psyfreak, 2017-10-21:
Same error

don, 2017-10-23:
Version 0.7.1 of phonegap-nfc now automatically adds the NFC framework to the project. It's possible this fixes the PhoneGap Build problem. This assumes that PhoneGap Build is compiling with Xcode 9.

markus-fischbacher, 2017-11-07:
Same problem here when building using Ionic Pro Package in the cloud.
Updated phonegap-nfc to 0.7.1 but still the same error present.

Any ideas?

```
[14:39:00]: ▸ In file included from /Users/ionic/builds/project-1/platforms/ios/Cook and Eat - order/Plugins/phonegap-nfc/NfcPlugin.m:7:
[14:39:00]: ▸ /Users/ionic/builds/project-1/platforms/ios/Cook and Eat - order/Plugins/phonegap-nfc/NfcPlugin.h:11:9: fatal error: 'CoreNFC/CoreNFC.h' file not found 
[14:39:00]: ▸ #import
[14:39:00]: ▸         ^ [14:39:00]: ▸ 1 error generated.
```

m-kikukawa, 2017-11-16:
Same error

jpgilchrist, 2017-12-11:
Same error here as well. I'm just running cordova, not phonegap build. However, for various (silly) reasons I need to run the build with xcode8.

jcesarmobile, 2018-02-01:
PGB started using Xcode 9 a few weeks ago, so this probably can be closed

don, 2018-04-18:
Closing based on https://github.com/chariotsolutions/phonegap-nfc/issues/281#issuecomment-362296136

closed 2018-04-18 by don